espressif/esp-matter #527: Moving files to their final destination failure (./install.sh)

sammichael25, 2023-07-20:
When attempting to run the install.sh, I get the undermentioned error.
Having manually checked, the directories referenced do not exist.

Host: Ubuntu 20.04 on WSL2 instance on Windows 10
uname -a:  5.15.90.1-microsoft-standard-WSL2
Python: 3.8.10

Prerequisite setup commands I ran:
ESP-IDF: 
```
sudo apt-get install git wget flex bison gperf python3 python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util 
libusb-1.0-0 
```

espressif/connectedhomeip: 
```
sudo apt-get install git gcc g++ pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev
```

From there the steps to [setup the repositories](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html#getting-the-repositories) was followed until the `./install.sh` command for esp-matter.

[image]

shubhamdp, 2023-07-24:
Can you try removing `.environment` directory and give it a one more try?
```
rm -rf connectedhomeip/connectedhomeip/.environment
./install.sh
```

shubhamdp, 2023-08-02:
@sammichael25 any updates on this ?

greenlightgroupofcompanies, 2023-08-02:
Hi @shubhamdp, yes I was able to get past this error by following this issue:
[https://github.com/project-chip/connectedhomeip/issues/26353](https://github.com/project-chip/connectedhomeip/issues/26353)
To be specific for anyone else that may be using WSL2 and encounter a similar error:

The armgcc package has 2 files with the same name, "QMemTags.html" and "qMemTags.html" in the same path. 
Since my checkout is in the Windows filesystem mount this doesn't work because Windows by default doesn't enable case sensitivity on files and directories. 

_For perspective, this prevents you from creating a file named "test.txt" in a directory that contains "TEST.txt"._

To fix the issue I had to ensure that the git clone was done on the "native" WSL FS or enable case-sensitivity on the parent folder before performing a clone of this repo. I opted to perform the following:

1. Take note of the repos parent directory.
2. Delete the current cloned repo.
3. Open a Powershell Admin window and execute the following command: 
`fsutil.exe file setCaseSensitiveInfo C:\Users\<Username>\<Parent-Directory> enable`
Replacing "\<Username\>" & "\<Parent-Directory\>" with your respective values.
Test that you can now create files named "test.txt" and "TEST.txt" in the parent directory.
4. Navigate to parent directory and re-clone repo again

The ./install.sh should no longer give an error about the duplicate files.

If the above doesn't work, try one of the other methods found here:
https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity

sammichael25, 2023-08-02:
Hi @shubhamdp, yes I was able to get past this error by following this issue:
[https://github.com/project-chip/connectedhomeip/issues/26353](https://github.com/project-chip/connectedhomeip/issues/26353)
To be specific for anyone else that may be using WSL2 and encounter a similar error:

The armgcc package has 2 files with the same name, "QMemTags.html" and "qMemTags.html" in the same path. 
Since my checkout is in the Windows filesystem mount this doesn't work because Windows by default doesn't enable case sensitivity on files and directories. 

_For perspective, this prevents you from creating a file named "test.txt" in a directory that contains "TEST.txt"._

To fix the issue I had to ensure that the git clone was done on the "native" WSL FS or enable case-sensitivity on the parent folder before performing a clone of this repo. I opted to perform the following:

1. Take note of the repos parent directory.
2. Delete the current cloned repo.
3. Open a Powershell Admin window and execute the following command: 
`fsutil.exe file setCaseSensitiveInfo C:\Users\<Username>\<Parent-Directory> enable`
Replacing "\<Username\>" & "\<Parent-Directory\>" with your respective values.
Test that you can now create files named "test.txt" and "TEST.txt" in the parent directory.
4. Navigate to parent directory and re-clone repo again

The ./install.sh should no longer give an error about the duplicate files.

If the above doesn't work, try one of the other methods found here:
https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity

closed 2023-08-02 by sammichael25