espressif/esp-matter #529: mode support help in mfg_tool

jonsmirl, 2023-07-21:
Something messed up in the help quoting...
~~~
  --supported-modes SUPPORTED_MODES [SUPPORTED_MODES ...]
                                                  List of supported modes, eg: mode1/label1/ep/"tagValue1\mfgCode,
                                                  tagValue2\mfgCode" mode2/label2/ep/"tagValue1\mfgCode, tagValue2\mfgCode"
                                                  mode3/label3/ep/"tagValue1\mfgCode, tagValue2\mfgCode"

~~~

I want to label three modes, skipping the Semantic Tags for now.  Three modes: "Switch", "Dimmer", "Smartbulb" on EP1
Like this, what about Attribute 0 - Description?
~~~
--supported-modes "1/Switch/1" "2/Dimmer/1" "3/Smartbulb/1"
~~~

jadhavrohit924, 2023-07-21:
@jonsmirl mfg_tool is used only for `SupportedModes` attribute. here is the [usage](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#usage).

> what about Attribute 0 - Description?

You can set `Description` right in the application.
```
cluster::mode_select::config conf;
strncpy(conf.mode_select_description, "description_for_mode", sizeof(conf.mode_select_description));
cluster::mode_select::create(endpoint, &conf, CLUSTER_FLAG_SERVER, ESP_MATTER_NONE_FEATURE_ID);
```

jonsmirl, 2023-07-21:
Description is flagged 'F' in the spec. I should have to use mfg_tool to set it.

[image]

jonsmirl, 2023-07-21:
This is accepted:
~~~
--supported-modes "1/Switch/1/1\1" "2/Dimmer/1/1\1" "3/Smartbulb/1/1\1"
~~~
but this isn't
~~~
--supported-modes "1/Switch/1" "2/Dimmer/1" "3/Smartbulb/1"
~~~
How does it work if I don't want any semantic tags?

jonsmirl, 2023-07-21:
What's going on here?
~~~
I (90161) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x0000FFFC is 0 **********
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42036a01  PS      : 0x00060430  A0      : 0x8202d428  A1      : 0x3fcb25b0  
0x42036a01: (anonymous namespace)::ModeSelectAttrAccess::Read(chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:75

A2      : 0x00000002  A3      : 0x3fcb2738  A4      : 0x3fcb2610  A5      : 0x00000050  
A6      : 0xfffffffd  A7      : 0x00000000  A8      : 0x820369f8  A9      : 0x3fcb2590  
A10     : 0x00000000  A11     : 0x3c1a3ff0  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4202df04  LEND    : 0x4202df1f  LCOUNT  : 0x00000000  
0x4202df04: MatterGetAccessPrivilegeForReadAttribute(unsigned long, unsigned long) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/privilege-storage.cpp:85

0x4202df1f: MatterGetAccessPrivilegeForReadAttribute(unsigned long, unsigned long) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/privilege-storage.cpp:91

Backtrace: 0x420369fe:0x3fcb25b0 0x4202d425:0x3fcb2610 0x4202e7b5:0x3fcb26a0 0x4202e8b5:0x3fcb26c0 0x4202eafb:0x3fcb27a0 0x4202ecd2:0x3fcb28c0 0x4202ed38:0x3fcb28f0 0x420de481:0x3fcb2910 0x420de491:0x3fcb2930 0x420e7eb1:0x3fcb2950 0x420e81a6:0x3fcb2970 0x420e81d9:0x3fcb29e0 0x40384ead:0x3fcb2a00
0x420369fe: (anonymous namespace)::ModeSelectAttrAccess::Read(chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:72 (discriminator 3)

0x4202d425: chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:450
 (inlined by) chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:577

0x4202e7b5: chip::app::reporting::Engine::RetrieveClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::AttributeReportIBs::Builder&, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:82

0x4202e8b5: chip::app::reporting::Engine::BuildSingleReportDataAttributeReportIBs(chip::app::ReportDataMessage::Builder&, chip::app::ReadHandler*, bool*, bool*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:185

0x4202eafb: chip::app::reporting::Engine::BuildAndSendSingleReportData(chip::app::ReadHandler*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:522

0x4202ecd2: chip::app::reporting::Engine::Run() at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:643

0x4202ed38: chip::app::reporting::Engine::Run(chip::System::Layer*, void*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:595

0x420de481: chip::System::TimerData::Callback::Invoke() const at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemTimer.h:61
 (inlined by) chip::System::TimerPool<chip::System::TimerList::Node>::Invoke(chip::System::TimerList::Node*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemTimer.h:237

0x420de491: chip::LambdaBridge::Initialize<chip::System::LayerImplFreeRTOS::ScheduleWork(void (*)(chip::System::Layer*, void*), void*)::{lambda()#1}>(chip::System::LayerImplFreeRTOS::ScheduleWork(void (*)(chip::System::Layer*, void*), void*)::{lambda()#1} const&)::{lambda(std::aligned_storage<24u, 4u>::type const&)#1}::_FUN(std::aligned_storage<24u, 4u>::type const) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplFreeRTOS.cpp:138
 (inlined by) operator() at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/LambdaBridge.h:39
 (inlined by) _FUN at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/LambdaBridge.h:39

[... cut ...]
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280

0x40384ead: vPortTaskWrapper at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: f157672e70d8c40f

Rebooting...

~~~

jonsmirl, 2023-07-21:
Making progress, now I can't link. I am using esp-matter/examples/common/blemesh_platform/platform/ESP32_custom
Maybe a missing link?
~~~
libdriver.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  esp-idf/mbedtls/libmbedtls.a  -Wl,--end-group && :
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ep.cpp.obj):(.literal._ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD5Ev[_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD5Ev]+0x0): undefined reference to `_ZTVN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerE'
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ep.cpp.obj):(.literal._ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD5Ev[_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD5Ev]+0x4): undefined reference to `_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManager5mSizeE'
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ep.cpp.obj):(.literal._ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD5Ev[_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD5Ev]+0x8): undefined reference to `_ZNK4chip3app8Clusters10ModeSelect27StaticSupportedModesManager18FreeSupportedModesEt'
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ep.cpp.obj):(.literal._ZN8Ep_Light4initEPjb+0x34): undefined reference to `_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManager17InitEndpointArrayEi'
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ep.cpp.obj): in function `_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManagerD2Ev':
/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/../../examples/platform/esp32/mode-support/static-supported-modes-manager.h:69: undefined reference to `_ZNK4chip3app8Clusters10ModeSelect27StaticSupportedModesManager18FreeSupportedModesEt'
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ep.cpp.obj): in function `_ZN10esp_matter7cluster11mode_select6configC4Ev':
/home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_cluster.h:514: undefined reference to `_ZN4chip3app8Clusters10ModeSelect27StaticSupportedModesManager17InitEndpointArrayEi'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jonsmirl/lowpan/build/log/idf_py_stderr_output_2831217 and /home/jonsmirl/lowpan/build/log/idf_py_stdout_output_2831217
jonsmirl@ares:~/lowpan$ 
~~~

jonsmirl, 2023-07-21:
So new crash....
~~~
I (23255) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x0000FFFC is 0 **********
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420e50ef  PS      : 0x00060230  A0      : 0x82036f6d  A1      : 0x3fcb25a0  
0x420e50ef: chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type::Encode(chip::TLV::TLVWriter&, chip::TLV::Tag) const at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp:11476

A2      : 0xbaad5678  A3      : 0x3fcb2874  A4      : 0x00000100  A5      : 0x00000000  
A6      : 0x00000100  A7      : 0x00000016  A8      : 0x820e50ec  A9      : 0x3fcb2580  
A10     : 0x00000000  A11     : 0x00000181  A12     : 0x00000100  A13     : 0x00000000  
A14     : 0x00000015  A15     : 0x3fcb25a8  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbaad5678  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x420e50ec:0x3fcb25a0 0x42036f6a:0x3fcb25d0 0x4202d929:0x3fcb2630 0x4202ecb9:0x3fcb26c0 0x4202edb9:0x3fcb26e0 0x4202efff:0x3fcb27c0 0x4202f1d6:0x3fcb28e0 0x4202f23c:0x3fcb2910 0x420de9f1:0x3fcb2930 0x420dea01:0x3fcb2950 0x420e8421:0x3fcb2970 0x420e8716:0x3fcb2990 0x420e8749:0x3fcb2a00 0x40384ead:0x3fcb2a20
0x420e50ec: chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type::Encode(chip::TLV::TLVWriter&, chip::TLV::Tag) const at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp:11475

0x42036f6a: chip::ChipError chip::app::DataModel::Encode<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type, (chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type*)0>(chip::TLV::TLVWriter&, chip::TLV::Tag, chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model/Encode.h:117
 (inlined by) chip::ChipError chip::app::AttributeReportBuilder::EncodeValue<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&, true>(chip::app::AttributeReportIBs::Builder&, chip::TLV::Tag, chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:82
 (inlined by) chip::ChipError chip::app::AttributeValueEncoder::EncodeListItem<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&>(chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:278
 (inlined by) chip::ChipError chip::app::AttributeValueEncoder::ListEncodeHelper::Encode<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&, true>(chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) const at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:127
 (inlined by) operator()<chip::app::AttributeValueEncoder::ListEncodeHelper> at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:87
 (inlined by) EncodeList<(anonymous namespace)::ModeSelectAttrAccess::Read(const chip::app::ConcreteReadAttributePath&, chip::app::AttributeValueEncoder&)::<lambda(const auto:5&)> > at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:227
 (inlined by) Read at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:82

0x4202d929: chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:450
 (inlined by) chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:577

0x4202ecb9: chip::app::reporting::Engine::RetrieveClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::AttributeReportIBs::Builder&, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:82

0x4202edb9: chip::app::reporting::Engine::BuildSingleReportDataAttributeReportIBs(chip::app::ReportDataMessage::Builder&, chip::app::ReadHandler*, bool*, bool*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:185

0x4202efff: chip::app::reporting::Engine::BuildAndSendSingleReportData(chip::app::ReadHandler*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:522

[... cut ...]
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/jonsmirl/lowpan/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280

0x40384ead: vPortTaskWrapper at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: 910a58c6a240b0f4

Rebooting...

~~~

jonsmirl, 2023-07-21:
I changed the array to 4 and it works.
~~~
    if (ep_dimmer) {
        mode_select::config_t mode_select_config;
        mode_select::create(epLight, &mode_select_config, CLUSTER_FLAG_SERVER, ESP_MATTER_NONE_FEATURE_ID);

        ModeSelect::StaticSupportedModesManager obj;
        obj.InitEndpointArray(4);
    }
~~~
Since my ModeSelect is on EP 1, maybe I could set this to 2? I have 17 EPs.
~~~
2023-07-21 16:07:31.971 12206-12206 RockerSetu...l$1$2$emit com.lowpan.m2                        D  Attrib [ModeSelectClusterModeOptionStruct {
                                                                                                    	label: Switch
                                                                                                    	mode: 1
                                                                                                    	semanticTags: [ModeSelectClusterSemanticTagStruct {
                                                                                                    	mfgCode: 1
                                                                                                    	value: 1
                                                                                                    }
                                                                                                    ]
                                                                                                    }
                                                                                                    , ModeSelectClusterModeOptionStruct {
                                                                                                    	label: Dimmer
                                                                                                    	mode: 2
                                                                                                    	semanticTags: [ModeSelectClusterSemanticTagStruct {
                                                                                                    	mfgCode: 1
                                                                                                    	value: 1
                                                                                                    }
                                                                                                    ]
                                                                                                    }
                                                                                                    , ModeSelectClusterModeOptionStruct {
                                                                                                    	label: Smartbulb
                                                                                                    	mode: 3
                                                                                                    	semanticTags: [ModeSelectClusterSemanticTagStruct {
                                                                                                    	mfgCode: 1
                                                                                                    	value: 1
                                                                                                    }
                                                                                                    ]
                                                                                                    }
                                                                                                    ]
~~~
I have those Semantic Tags in there that I don't want.

jadhavrohit924, 2023-07-23:
> I have those Semantic Tags in there that I don't want.

Semantic tags are mandatory in spec. We can’t skip it.

> Description is flagged 'F' in the spec. I should have to use mfg_tool to set it.

We don’t use mfg_tool for all the things that are fixed. Having flagged `F` means it should be change at all in the lifetime of program.

jonsmirl, 2023-07-23:
Support for semantic tags is mandatory, but my list could be empty.

jonsmirl, 2023-07-23:
Also static-supported-modes-manager.cpp, static-supported-modes-manager.h need to be fixed up in esp-matter/examples/common/blemesh_platform

jadhavrohit924, 2023-07-25:
> Support for semantic tags is mandatory, but my list could be empty. How do I make an empty list?

From spec list cannot be empty.
[image]

> Also static-supported-modes-manager.cpp, static-supported-modes-manager.h need to be fixed up in esp-matter/examples/common/blemesh_platform

We will look into it.

jonsmirl, 2023-07-26:
> > Support for semantic tags is mandatory, but my list could be empty. How do I make an empty list?
> 
> From spec list cannot be empty. [image]
> 

BTW, this would say "MUST have" instead of "MAY have" if the tags were mandatory.

jadhavrohit924, 2023-07-28:
> I got it to link by copying into my project. esp-matter/examples/common/blemesh_platform/platform/ESP32_custom needs to be fixed up to compile without me copying.

@jonsmirl I don’t understand this part. I could able to compile blemesh_bridge app successfully. Could you please explain me more where you are facing compilation failure ?

jadhavrohit924, 2023-07-28:
FYI, semantic tags can be empty, we are fixing it. Thanks for pointing out.

jonsmirl, 2023-07-28:
> @jonsmirl I don’t understand this part. I could able to compile blemesh_bridge app successfully. Could you please explain me more where you are facing compilation failure ?

I am using the both the BLE Bridge and Mode Select.   Since using BLE Bridge forces me to use esp-matter/examples/common/blemesh_platform/platform/ESP32_custom, the esp-matter/examples/common/blemesh_platform/platform/ESP32_custom has to still work while ModeSelect is in use. 

Links to 
connectedhomeip/connectedhomeip/examples/platform/esp32/mode-support/static-supported-modes-manager.cpp  and connectedhomeip/connectedhomeip/examples/platform/esp32/mode-support/static-supported-modes-manager.h have to be present in esp-matter/examples/common/blemesh_platform/platform/ESP32_custom

In other words, temporarily add ModeSelect into the BLE Bridge example add and then get it to compile.  I tried adding the links, so something more is needed to make it work.

jadhavrohit924, 2023-07-28:
Okay. I tried to compile BLE Briged app with mode select and it compiled successfully. Which esp-matter and connectedhomeip you are using?

jonsmirl, 2023-07-28:
Did you add this?

        ModeSelect::StaticSupportedModesManager obj;
        obj.InitEndpointArray(4);

Then it won't link without: connectedhomeip/connectedhomeip/examples/platform/esp32/mode-support/static-supported-modes-manager.cpp and connectedhomeip/connectedhomeip/examples/platform/esp32/mode-support/static-supported-modes-manager.h

jonsmirl, 2023-07-28:
Should this be in my code or in esp-matter? Currently I have to add it.
~~~
    ModeSelect::StaticSupportedModesManager obj;
    obj.InitEndpointArray(4);
~~~

closed 2023-08-04 by dhrishi