espressif/esp-matter #530: Use mfg_tool to generate factory firmware flash to esp32, HomeAPP does not take effect

vaemc, 2023-07-21:
Hi， I generated the factory firmware using this [command](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#generate-a-factory-partition) and flashed it to ESP32, but it doesn't work. What could be the reason?

 ./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --serial-num 12345678 --hw-ver-str 2.1 --product-name Doit --pai \
   -k /home/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
   -c /home/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
   -cd /home/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der

Are there any additional configurations that need to be done?

[image]

jonsmirl, 2023-07-21:
Did you write the partition to the device? For example...
~~~
esptool.py -p /dev/ttyACM0 write_flash 0x10000 ab331b1f-95a2-4e97-968b-11a17a4584a8-partition.bin
~~~

vaemc, 2023-07-22:
> Did you write the partition to the device? For example...
> 
> ```
> esptool.py -p /dev/ttyACM0 write_flash 0x10000 ab331b1f-95a2-4e97-968b-11a17a4584a8-partition.bin
> ```

Yes, I have written the firmware to the device and paired it with Home.

jonsmirl, 2023-07-22:
I am using this...
~~~
./mfg_tool.py -cn "Test" -v 0xFFF1 -p 0x8002 --pai \
-k ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd ../../connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Lowpan" --product-name "Dimmer" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840 \
--fixed-labels "1/name/Main" "2/name/BTN-2" "3/name/BTN-3" "4/name/BTN-4" "5/name/BTN-5" "6/name/BTN-6" "7/name/BTN-7"  "8/name/BTN-8" \
--supported-modes "0/Switch/1/1\1" "1/Dimmer/1/1\1" "2/Smartbulb/1/1\1"
~~~

vaemc, 2023-07-22:
> I am using this...
>
> ```
> ./mfg_tool.py -cn "Test" -v 0xFFF1 -p 0x8002 --pai \
> -k ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
> -c [...]

Thank you for your reply. The command you sent still has no effect. Do I need to open any options in menuconfig?

vaemc, 2023-07-22:
😀Solved, need to select Device Instance Info - Factory
[image]

w

vaemc, 2023-07-22:
But why is my serial number not displayed?

[image]

jonsmirl, 2023-07-22:
Maybe HomeApp is not showing this live? Did it record it at commissioning time and you need to recommission to update it?

sayondeep, 2023-07-25:
@vaemc You have correctly set the Device Instance Info - Factory

Serial-number is an optional attribute of Basic Information Cluster. We need to add it on the application side for it to take effect.

The below mentioned lines can be used as reference for creating it.
```
    endpoint_t * basic_info_endpoint = endpoint::get(node,0);
    if(!basic_info_endpoint)
        printf("Could not fetch Basic Information Endpoint\n");
    else
    {
        printf("Basic Information Endpoint fetched....adding serial number\n");
        cluster_t *basic_info_cluster =  cluster::get(basic_info_endpoint, BasicInformation::Id);
        cluster::basic_information::attribute::create_serial_number(basic_info_cluster,NULL,0);
    }
```

vaemc, 2023-07-26:
@sayondeep Thank you very much for your response, which solved my problem！

closed 2023-07-26 by vaemc