espressif/esp-matter #531: logic around unique-id from BasicInformation cluster

albrownwood, 2023-07-21:
The Matter specification states that the unique-id shall be updated on each factory reset among other requirements.
Is the esp-matter stack automatically handling the initial creation and update of the unique-id or is this the responsibility of the application layer to manage it?
If handled by the esp-matter stack what is the best way to obtain it from the application layer?

shubhamdp, 2023-07-22:
To get/set/generate the unique-id, you will need an instance of [configuration manager](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/ConfigurationManager.h#L185) and few APIs from [Configuration Manager](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/internal/GenericConfigurationManagerImpl.ipp#L499-L525).

For the first part of question, Unique-id is stored in `chip-config` [namespace](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32Config.cpp#L91), and this namespace gets erased on [factory reset](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ConfigurationManagerImpl.cpp#L382). A new unique-id will be [generated](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/internal/GenericConfigurationManagerImpl.ipp#L264) if not present in the NVS.

albrownwood, 2023-07-22:
thanks @shubhamdp very clear and this works well.

closed 2023-07-22 by albrownwood