espressif/esp-matter #532: Development matter with esp

wch188, 2023-07-22:
I would like to ask if the product development using esp-matter, such as the chip using esp32c3, can skip the official matter certification process? Or when developing with an ESP-Matter-enabled chip, what is the process from product development to market? If you can skip the matter certification process, what steps do you need to develop a good product to market?

wch188, 2023-07-22:
In addition, when the app of other platforms is connected to the device, can the icon of the relevant product be placed on the interface? Is there any interface that can change the manufacturer information when the device broadcasts to the outside world? (No interface similar to setting the supplier ID can be found in the routine, nor can the menu configuration be seen)? Does this need to be negotiated with the corresponding platform and then let them add? Or when other apps connect, the interface will only display the same preset graphics (like Apple's linght routine?).

jonsmirl, 2023-07-22:
Google lets you upload product images. When they see your products, they will use those images. Don't know about Apple and Amazon.

You have to do a low level of certification, I believe Espressif offers it as a service. You need to get a vendor ID assigned from Matter so that you can put OTA update images into the system. The lowest possible fee is $2500 plus $500/yr.

You can run locally using the test Vendor and test Product ID. Everything will work except OTA.  You can't ship with those IDs because every developer is using the same test IDs.

wch188, 2023-07-24:
[image]

So where do I put my product images so Google can see them?

If the company already has a manufacturer ID (not assigned by matter) and does not intend to use the ota functionality of matter, can it just use the current ID? And after applying for the manufacturer ID from matter, can the firmware upgraded by ota be placed on the server related to matter, or is it still placed on its own server?

Where should the manufacturer ID and product ID be changed? I tried to find it in the light routine code and menu configuration, but I did not see the manufacturer ID and product ID printed on the log, and many of the code in the log are chip logs, which seem to be packaged and cannot find them.

jonsmirl, 2023-07-24:
Product images are uploaded via the Google Home console
https://console.home.google.com/projects

You can only use Matter assigned vendor IDs. There is a phase of bringing your devices on to the network call attestation. During that phase Matter looks for counterfeit products using public key encryption. Your product will be marked as unauthorized and not allowed onto the network.

The test vendor and product IDs avoid this check, but everyone is using the same test IDs.

You set the Vendor (manufacturer ID) and Product ID via ESP menuconfig

wch188, 2023-07-24:
Ok, thank you. By the way, will the product ID be given together when applying for the manufacturer ID? And when the product is developed in the above way, does it need to undergo any testing or certification before it can be marketed?

wch188, 2023-07-24:
[image]
I'll look for them later and see if I missed something and I didn't find them

jonsmirl, 2023-07-24:
You can pick any product ID you want from 0-65535

wch188, 2023-07-24:
Is there any requirement for selecting the discriminator parameter when chip-tool is used for ble network distribution? I have seen the value 3841 used while looking at some examples.

wch188, 2023-07-24:
Is there any requirement for selecting the discriminator parameter when chip-tool is used for ble network distribution? I have seen the value 3841 used while looking at some examples. Is this value used only when connecting to the matter network? Still in the stage of bluetooth use, equipment ble broadcast packets of data parameters only this 0 x0201060b16f6ff00000ff1ff008000, is inside a F00 (3840)?

shubhamdp, 2023-07-24:
@wch188 Hope the [production consideration guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/production.html) would be able to answer your question.

wch188, 2023-07-24:
I checked, but there is no mention of the discriminator value range or where to get it, and the production engineering image directive only mentions using the existing discriminator (3840).

shubhamdp, 2023-07-24:
Discriminator is 12 bit value. I'd say for more technicals you can check the Matter Specification Section 5.1. Onboarding Payload. You can download the specification from [CSA's official site](https://csa-iot.org/developer-resource/specifications-download-request/).

wch188, 2023-07-24:
Ok Thank you.

closed 2023-07-24 by wch188