espressif/esp-matter #535: unable to write in the factory partition (fctry)

albrownwood, 2023-07-25:
I am trying to write some calibration values in the NVS factory partition (fctry), the goal is for these data to persist after a factory reset.

when I try to obtain a handle on that partition using nvs_open_from_partition in NVS_READWRITE mode, I get a ESP_ERR_NVS_PART_NOT_FOUND error. Are there specific restrictions on that partition that prevent me from getting a handle?

shubhamdp, 2023-07-25:
@albrownwood It would be good if you share a patch or code snippet for what you are trying to do.

But, I think you may be doing this before calling `esp_matter::start()`, because there is no way you would get error after that. And if you want to do that before `esp_matter::start()` call, then you will have to initialize the NVS partition using [`nvs_flash_init_partition()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#_CPPv424nvs_flash_init_partitionPKc) API.

API definition for [nvs_open_from_partition](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#_CPPv423nvs_open_from_partitionPKcPKc15nvs_open_mode_tP12nvs_handle_t).

Please see the below snippet, exactly similar what you are looking for https://github.com/espressif/esp-matter/blob/8801d8e930613e49e5c139d70bd5b815105b9a7a/examples/generic_switch/main/app_main.cpp#L199-L213

albrownwood, 2023-07-26:
thanks for the help @shubhamdp, in my sequence of operations the nvs_flash_init_partition() was not called for the factory partition.

closed 2023-07-26 by albrownwood