espressif/esp-matter #537: Crash during shutting down Matter

8TRobi8, 2023-07-25:
I am trying to shut down the Matter core but I always got a crash.

This is what I try to do:
    chip::DeviceLayer::PlatformMgr().HandleServerShuttingDown();
    chip::Server::GetInstance().DispatchShutDownAndStopEventLoop();

The first function call is successful but the second one crashes all the time with the following information:

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4012785c: ffeca502 0000f01d 81006136
0x4012785c: chip::DeviceLayer::PlatformManager::RunEventLoop() at connectedhomeip/src/include/platform/PlatformManager.h:362
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257

Core  1 register dump:
PC      : 0x40127860  PS      : 0x00060430  A0      : 0x00000000  A1      : 0x3ffe6920
0x40127860: chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:261

A2      : 0x3ffc0248  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80127860  A9      : 0x3ffe68b0
A10     : 0x3ffc0248  A11     : 0x3ffc0248  A12     : 0x3ffc0269  A13     : 0x00000000
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400820bd  LEND    : 0x400820c5  LCOUNT  : 0x00000027
0x400820bd: esp_timer_impl_get_counter_reg at esp-idf/v4.4.2/components/esp_timer/src/esp_timer_impl_lac.c:118 (discriminator 2)

0x400820c5: esp_timer_impl_get_counter_reg at esp-idf/v4.4.2/components/esp_timer/src/esp_timer_impl_lac.c:128 (discriminator 1)

Backtrace:0x4012785d:0x3ffe6920
0x4012785d: chip::DeviceLayer::PlatformManager::RunEventLoop() at connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:362
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257

Could you help me out how to solve this issue?

shubhamdp, 2023-08-02:
@8TRobi8 Thanks for reporting, will look into it.

sayondeep, 2024-02-05:
@8TRobi8 can you please try using `chip::Server::GetInstance().Shutdown()` to shutdown matter.

dhrishi, 2024-08-09:
Closing this now

closed 2024-08-09 by dhrishi

bilalmalik76, 2024-12-11:
This is also causing crash:
chip::Server::GetInstance().Shutdown()