espressif/esp-matter #538: Understanding of some of the CHIP Device side Configurations

aggaddam, 2023-07-25:
I want to understand the below configurations in CHIP Device SDK. 

```
const ESP32Config::Key ESP32Config::kConfigKey_SetupPinCode            = { kConfigNamespace_ChipFactory, "pin-code" };
const ESP32Config::Key ESP32Config::kConfigKey_MfrDeviceId                = { kConfigNamespace_ChipFactory, "device-id" };
const ESP32Config::Key ESP32Config::kConfigKey_MfrDeviceCert            = { kConfigNamespace_ChipFactory, "device-cert" };
const ESP32Config::Key ESP32Config::kConfigKey_MfrDeviceICACerts     = { kConfigNamespace_ChipFactory, "device-ca-certs" };
const ESP32Config::Key ESP32Config::kConfigKey_MfrDevicePrivateKey  = { kConfigNamespace_ChipFactory, "device-key" };
const ESP32Config::Key ESP32Config::kConfigKey_SupportedCalTypes    = { kConfigNamespace_ChipFactory, "cal-types" };
const ESP32Config::Key ESP32Config::kConfigKey_SupportedLocaleSize = { kConfigNamespace_ChipFactory, "locale-sz" };
const ESP32Config::Key ESP32Config::kConfigKey_LocationCapability      = { kConfigNamespace_ChipFactory, "loc-capability" };
```

1) Can someone explain what are these configurations for? Is it mandatory to update these configurations during the device manufacturing process? 
2) Is  kConfigKey_SetupPinCode refer to the passcode value used during the commissioning process or a different one? If it refers to passcode, do we need to store the passcode on the device side or it is optional?

shubhamdp, 2023-07-25:
There could be some configurations that can be stored in non volatile memory, the ones you listed are a few. These belongs to `chip-factory` namespace, and should be updated during the device manufacturing process. These won't be erased on factory reset and are expected to stay the same throughout the lifetime of device.

Matter specification has various clusters, and the listed configurations are the attributes from there. One will have to check with the specification which ones are mandatory or not.

The ones that you have listed seems to be from basic information and localization configuration. Also, you will find some configurations are not used at all, eg: "device-cert".

Regarding setup pincode, pincode is not supposed to be stored on the device but the verifier corresponding to that pincode (please see Matter specification section 5.1.7. Generation of the Passcode). I think this value in here makes things confusing and looks like we need to remove it.

shubhamdp, 2023-08-02:
@aggaddam If your queries are addressed can you please close the issue.

closed 2023-08-02 by aggaddam