espressif/esp-matter #540: File Transfer (not-OTA)

Gonta01, 2023-07-26:
Is there any ability to transfer files between nodes? Matter specifies BDX protocol for this purpose. Are there any BDX usage examples? What is needed to implement it properly?

jonsmirl, 2023-07-26:
Diagnostics Logs Cluster uses BDX to send files to the controller, is there any way for the controller to send files to the device outside of OTA?

shubhamdp, 2023-07-27:
@Gonta01 I haven't seen any direct bdx example in connectedhomeip repository. You may have to look into the ota-provider and ota-requestor examples for that.

But, As far as I can tell, you will have to register the handler for unsolicited message handler for bdx protocol for receiving them. Construct the BDX message and use the exchange context for sending the it.

This [TransferFacilitator.h](https://github.com/project-chip/connectedhomeip/blob/master/src/protocols/bdx/TransferFacilitator.h) can be used as well.

Gonta01, 2023-07-27:
@shubhamdp thank you for reply! Custom cluster (similar to ota-provider/ota-requestor/diagnostics logs) must be created to transfer some files from controller to device?

jonsmirl, 2023-07-27:
There should be two implementations already in the system: Diagnostics Logs Cluster  which sends files. OTA which receives files. I would suggest working with those first before trying to build a custom cluster.

Also note, there is nothing stopping you from just opening up a socket and talking to whoever you want. I have a couple of devices here which are ignoring the Matter OTA system and they use their own apps to update via a private protocol. Of course that also bypasses Matter security.

I do think it is possible to hack OTA into transferring files. My idea, which I have not tried yet, it to make the user's phone into a temporary OTA provider. Then request an OTA from it filling in the metadata field of the request.  The phone would see that metadata and not send an OTA image back, instead it would send another file. Then the code in the device needs to differentiate between receiving an OTA file and a file for a different purpose.

Gonta01, 2023-07-27:
@jonsmirl thank you for advices!

shubhamdp, 2023-08-02:
@Gonta01 If your queries are addressed can you please close this issue.

closed 2023-08-02 by Gonta01