espressif/esp-matter #541: Generation of DAC key with in the device and update DAC private key to factory nvs partition

aggaddam, 2023-07-26:
I am exploring the option to generate DAC private key within the device itself (ESP32C3) for security and update the DAC private key into the factory partition using write API and will send CSR to our cloud to get the DAC certificate. 

I am a little confused about which APIs to be used here. I can see two options 1) from Chip SDK and 2) Esp secure cert-manager.

Here is the sample code. With this, I can get the pub key but not the private key. Is there a way I can get a private key from the helper function?

```
Crypto::P256Keypair dac_Key;
if (CHIP_NO_ERROR != dac_Key.Initialize(Crypto::ECPKeyTarget::ECDSA))
{
    return ESP_FAIL;
}

dac_Key.Pubkey();  // Getting pub key 
```

wqx6, 2023-07-27:
You could use `P256Keypair::Serialize()` to get a P256SerializedKeypair data buffer, the private key will be `serializedKeyPair + kP256_PublicKey_Length`.

aggaddam, 2023-07-27:
@wqx6 

serializedKeyPair + kP256_PublicKey_Length -> Is this gives raw 32 bytes of private key data?

shubhamdp, 2023-07-27:
@aggaddam I would suggest that you use the APIs from connectedhomeip SDK, please take a look at [`class P256Keypair`](https://github.com/project-chip/connectedhomeip/blob/master/src/crypto/CHIPCryptoPAL.h#L487).

Regarding length: PTAL at https://github.com/project-chip/connectedhomeip/blob/master/src/crypto/CHIPCryptoPAL.h#L83-L84

And format for serialized key: https://github.com/project-chip/connectedhomeip/blob/master/src/crypto/CHIPCryptoPAL.h#L463

aggaddam, 2023-07-27:
s@shubhamdp Thanks for providing more info. I am exploring things in chip SDK and how it's implemented.

I am hoping I did the correct way in getting the private key and public key using chip SDK API's

```
esp_err_t matter_crypto_util_gen_P256_keypair(void)
{
    Crypto::P256Keypair dac_Key;
    if (CHIP_NO_ERROR != dac_Key.Initialize(Crypto::ECPKeyTarget::ECDSA))
    {
        return ESP_FAIL;
    }

    // Create a P256SerializedKeypair structure to hold the serialised key data
    Crypto::P256SerializedKeypair serialized_key_pair;
    if (CHIP_NO_ERROR != dac_Key.Serialize(serialized_key_pair))
    {
        return ESP_FAIL;
    }

    print_hexdata(serialized_key_pair.Bytes(), Crypto::kP256_PublicKey_Length);
    print_hexdata(serialized_key_pair.Bytes(), Crypto::kP256_PrivateKey_Length);

    return (ESP_OK);
}
   
```

shubhamdp, 2023-08-02:
@aggaddam I was going to suggest the same, sign/verify to validate your changes. If Sign and Verify works then I see these changes are good to go.

> Do I need to store the raw private and public keys in the factory nvs partition or .pem files?

You can store however you would like. You can store in any format, even the whole serialized buffer for conveniently passing the same to deserialize it. "PEM" (base64 encoded format) may not be useful here, as you would have to base64 encode/decode when storing/getting it.

Surely you can use `dac-key` and `dac-pub-key` keys to store the key into NVS, but I'd suggest you store it somewhere else where only you knows the place, just to keep it more secure. Also, avoid print it.

I am hoping you will implement the [DeviceAttestationCredsProvider](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/DeviceAttestationCredsProvider.h) interface, or inherit the one that [ESP32 has implemented](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.h#L56) for this purpose.

shubhamdp, 2023-08-10:
@aggaddam If the issue is resolved can we please close this.

dhrishi, 2023-08-12:
Closing this now

closed 2023-08-12 by dhrishi