espressif/esp-matter #542: Is it possible to use Espressif's provisioning solution with Matter commissioning?

law-ko, 2023-07-26:
Hello,

Is it possible that the [Espressif's provisioning solution](https://github.com/espressif/esp-idf/tree/d2471b11e78fb0af612dfa045255ac7fe497bea8/examples/provisioning/wifi_prov_mgr) can coexist and work with Matter's commissioning at the same time?

Thank you.

dhrishi, 2023-07-27:
@law-ko Matter commissioning involves setting up the Wi-Fi network. The same is done through the Wi-Fi prov manager. Is there any specific reason you need that as it will be automatically taken care of?

law-ko, 2023-07-27:
We want to be backward compatible with our previous app design. And since Matter's commissioning uses NimBLE, when initializing the esp provisioning service, it will crash due to NimBLE GATT server is already being started by Matter.

dhrishi, 2023-07-27:
@law-ko In that case you will use the on-network commissioning for Matter? One option I can think of now is using the external platform, as done in the BLE Mesh to Matter bridge example. Then you can make customized changes as per your requirements. Details [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#creating-the-external-platform-directory)

law-ko, 2023-07-27:
@dhrishi If we use Matter's provisioning method, is it possible for the app to get the hostname or the IP address of the provisioned device?

dhrishi, 2023-07-27:
@law-ko Yes, that's possible. Once the device gets commissioned, it starts operational advertisement using mDNS. Just to try, you can find all such devices by running the following command on a machine in the same network
 `avahi-browse _matter._tcp -r` from Linux based machine
OR
`dns-sd -B _matter._tcp` from a MacOS

law-ko, 2023-07-27:
I think many Android phones still do not support mDNS. Any way to find the device after commissioning for Android?

dhrishi, 2023-07-28:
Yes, mDNS does work on Android. @law-ko Please close this issue if your questions are answered

closed 2023-07-31 by dhrishi

law-ko, 2023-08-07:
@dhrishi Can we open SoftAP for provisioning (not Matter softAP) on top of Matter commissioning? And once the SoftAP has successfully provisioned to ESP32, the ESP32 will send commissioning complete to Matter service.

dhrishi, 2023-08-07:
@law-ko Giving the device IP connectivity (connecting to Wi-Fi/Thread network) is a part of Matter commissioning. If you want to do it through any other mechanism, like SoftAP, then once the device gets IP connectivity then you can perform on-network commissioning. In that case, PASE session will be established on the IP network itself as opposed to BLE in the default cases

law-ko, 2023-08-07:
@dhrishi Can you provide an example on how to turn on SoftAP and establishing the PASE session on the IP network?

dhrishi, 2023-08-07:
@law-ko The logic around sending the Wi-Fi credentials is something you will have to take care of.You can refer Wi-Fi provisioning manager example in ESP-IDF. 

To quickly try the same usecase, you can use a CLI command to connect to the Wi-Fi network in the esp-matter example:

```
matter esp wifi connect <SSID> <Passphrase>
```

Once done, you can do onnetwork commissioning using chip-tool:

```
chip-tool pairing onnetwork node-id setup-pin-code
```

law-ko, 2023-08-07:
@dhrishi How do we turn on softAP on top of Matter service?

Also, can you provide the function that handles this CLI command?
`matter esp wifi connect <SSID> <Passphrase>`

Thank you

law-ko, 2023-08-09:
@dhrishi Manually setting up SoftAP after esp_matter::start seemed to have issue. 

```
    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "Matter",
            .ssid_len = strlen("Matter"),
            .channel = 1,
            .max_connection = 4,
            .pmf_cfg = {
                    .required = false,
            },
        },
    };
    if (1) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config((wifi_interface_t) 1, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_LOGW(TAG, "WiFi AP started");
    ```
    
 I am able to see the network on the Wi-Fi list but unable to connect to it. 
 
 Here is the output log: 
```
I (66790) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (66790) wifi:station: e6:6d:19:74:69:9b join, AID=2, bgn, 20
W (66790) wifi:m f assoc rsp l=111

HEAP left: 1983260 of 304652 (320KB) 
I (67160) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (67160) wifi:station: e6:6d:19:74:69:9b join, AID=2, bgn, 20
W (67160) wifi:m f assoc rsp l=111

I (67580) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (67580) wifi:station: e6:6d:19:74:69:9b join, AID=2, bgn, 20
W (67580) wifi:m f assoc rsp l=111

HEAP left: 1983260 of 304652 (320KB) 
I (67930) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (67930) wifi:station: e6:6d:19:74:69:9b join, AID=2, bgn, 20
W (67930) wifi:m f assoc rsp l=111

HEAP left: 1983260 of 304652 (320KB) 
I (68280) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (68280) wifi:station: e6:6d:19:74:69:9b join, AID=2, bgn, 20
W (68280) wifi:m f assoc rsp l=111
```