espressif/esp-matter #543: ESP32-C3 Matter Commissioning issue

aashikareliya, 2023-07-27:
**I am using ESP32-C3 Dev Kit and tried to do matter commissioning. ESP32 is unable to fetch the wifi credentials somehow. But it is possible same with mobile internet using hotspot. What can be the reasons? Below are the logs for the issue faced during commissioning.** 

W (78384) wifi:Haven't to connect to a suitable AP now!
E (78394) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (81734) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (81744) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (81754) wifi:Haven't to connect to a suitable AP now!
I (81754) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (81764) chip[DL]: Next WiFi station reconnect in 100 ms
I (81764) chip[DL]: Done driving station state, nothing else to do...
W (81774) wifi:Haven't to connect to a suitable AP now!
E (81774) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (81794) wifi:Haven't to connect to a suitable AP now!
E (81794) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (81864) wifi:Haven't to connect to a suitable AP now!
I (81864) chip[DL]: Attempting to connect WiFi station interface
I (81864) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (81884) chip[DL]: Done driving station state, nothing else to do...
W (81884) wifi:Haven't to connect to a suitable AP now!
E (81894) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (81974) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (81984) wifi:state: init -> auth (b0)
I (81984) wifi:state: auth -> assoc (0)
I (81994) wifi:state: assoc -> run (10)
I (85394) wifi:state: run -> init (fc0)
I (85394) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (85394) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (85394) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (85404) wifi:Haven't to connect to a suitable AP now!
I (85414) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (85424) chip[DL]: Next WiFi station reconnect in 100 ms
I (85434) chip[DL]: Done driving station state, nothing else to do...
W (85434) wifi:Haven't to connect to a suitable AP now!
E (85444) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (85444) wifi:Haven't to connect to a suitable AP now!
E (85464) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (85534) wifi:Haven't to connect to a suitable AP now!
I (85534) chip[DL]: Attempting to connect WiFi station interface
I (85534) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (85544) chip[DL]: Done driving station state, nothing else to do...
W (85554) wifi:Haven't to connect to a suitable AP now!
E (85564) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F

shubhamdp, 2023-07-27:
@aashikareliya Logs says that it is unable to connect, please make sure you have provided the correct SSID and Passphrase. Also, make sure the Wi-Fi AP is using 2.4 GHz band.

Also, I would suggest you to please give a try to [esp-idf/wifi/station](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) example.

If you still facing the problem, please attach(do not paste) logs from the device, if you are using the chip-tool as a controller also attach logs from chip-tool as well.

aashikareliya, 2023-07-27:
@shubhamdp 
esp-idf/wifi/station is working perfect.
my wifi router supports 2.4GHz and 5GHz both.
Is it possible that we face issue because wifi contains '@' symbols in it.
we tries it using the command : matter esp wifi connect <ssid> <password>

1. Is there any different solution ?
2. are there any other ways?

shubhamdp, 2023-07-27:
> my wifi router supports 2.4GHz and 5GHz both.

I hope they don't have the exactly same names which would conflict ESP32 into connecting to a 5GHz network.

> Is it possible that we face issue because wifi contains '@' symbols in it.

`@` is a valid character in an SSID, so no problem there.

You can use the already available commissioners for commissioning the device. `chip-tool` is a host based tool which can be used to commission the device. Please check the guide for [Commissioning and Control](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control).

jonsmirl, 2023-07-27:
My system works just fine with the same SSID/PW on 5Ghz and 2.4Ghz. Apple/Google give the SSID/PW to the ESP and the ESP happily connects using 2.4Ghz.

aashikareliya, 2023-07-28:
@shubhamdp Is there any other alternative for the android chiptool application?

shubhamdp, 2023-07-28:
Google Home Sample App for Android: https://github.com/google-home/sample-apps-for-matter-android/releases

aashikareliya, 2023-07-28:
@shubhamdp any other than GHSAFM?

shubhamdp, 2023-07-28:
Please take a look at https://github.com/espressif/esp-matter/issues/544#issuecomment-1653422316

aashikareliya, 2023-07-28:
> Please take a look at [#544 (comment)](https://github.com/espressif/esp-matter/issues/544#issuecomment-1653422316)

Thank You

shubhamdp, 2023-07-28:
@aashikareliya Can you please close the issue if your question is answered.

closed 2023-07-28 by aashikareliya