espressif/esp-matter #544: other mobile app for commissioning in ESP

gautam1joshi, 2023-07-27:
hello,
i want to commission the ESP matter device to network but chip-tool for android not working properly,
so what other APP for mobile i can use in Android?

shubhamdp, 2023-07-27:
@gautam1joshi You can use the google home sample app: https://github.com/google-home/sample-apps-for-matter-android/releases

gautam1joshi, 2023-07-27:
yes i try @shubhamdp  that but i don't see other feature except On/Off , so can i have to edit the app or do something?

shubhamdp, 2023-07-27:
This is just a sample app which tests commissioning. The other apps that I know for Android is Google Home App and Amazon Alexa app. But, you will need a home hub for that, Google nest hub or Amazon Echo.

Yeah, you may have to add start editing that stuff and create your own app. Or you can use the host based test tool, [chip-tool](https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool).

gautam1joshi, 2023-07-27:
okay, thank you.

closed 2023-07-27 by gautam1joshi