espressif/esp-matter #545: ESP32: matter build issue for ESP32 C6 DevKitC 1

manamohan1, 2023-07-28:
Hi,
        I am getting the following error message during the idf.py build
==========================================

/home/manamohana/ESP/esp-matter/examples/light/build/config/sdkconfig.h:452:37: error: invalid conversion from 'int' to 'uart_port_t' [-fpermissive]
  452 | #define CONFIG_ESP_CONSOLE_UART_NUM 0
      |                                     ^
      |                                     |
      |                                     int
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/shell/streamer_esp32.cpp:61:35: note: in expansion of macro 'CONFIG_ESP_CONSOLE_UART_NUM'
[build_error.txt](https://github.com/espressif/esp-matter/files/12189921/build_error.txt)

   61 |     if (!uart_is_driver_installed(CONFIG_ESP_CONSOLE_UART_NUM))
========================================================
Please find the attached file for complete console log.

shubhamdp, 2023-07-28:
I see an error setting up target. You will have to clean the stale files manually and give it another try. Please do `rm -rf build sdkconfig` and then do `idf.py set-target esp32c6` and `idf.py build`.

```
manamohana@manamohana-HP-Laptop-15s-fq2xxx:~/ESP/esp-matter/examples/light$ idf.py set-target esp32c6
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/home/manamohana/ESP/esp-matter/examples/light/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete     the directory manually to 'clean' it.
```
Due to this error the build target is now ESP32 and not C6.

Also, looks like you are using esp-idf master branch, and we have not tested esp-matter on master branch. I would suggest you to please use the latest stable v5.1 release. i.e. v5.1 tag.

dhrishi, 2023-07-28:
@manamohan1 In addition to @shubhamdp 's comment - In the attached log, your `install.sh` command has failed for `pkg-config`. Please refer this section in the documentation and install the pre-requisites for ESP-IDF and Matter: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup

marciogranzotto, 2023-07-28:
@manamohan1 you need to typecast it on the connectedhomeip repository
[image]

manamohan1, 2023-08-01:
I did "git checkout ea5e0ff" and this issue resolved for me.
==========================
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf; git checkout ea5e0ff; git submodule update --init --recursive;
./install.sh
==============================

closed 2023-08-02 by dhrishi