espressif/esp-matter #546: BLE Advertisement in ESP32-C3

aashikareliya, 2023-07-28:
I am using ESP32-CR DevKit and trying to do BLE Commissioning and matter binding. For that how to start Advertisement ?

dhrishi, 2023-07-28:
@aashikareliya What are you actually trying to do? What changes have you made in the standard examples? Also, are there any logs you can share? Please provide all the relevant details and then re-open the issue

FYI: You can commission [Light](https://github.com/espressif/esp-matter/tree/main/examples/light) and [light switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) examples and then create binding between them. See [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#2-post-commissioning-setup)

closed 2023-07-28 by dhrishi

aashikareliya, 2023-07-31:
I am trying to do ble commissioning with other manufacturer device which is matter enabled and support BLE 5.3. I am using Light Bulb example in ESP32-C3 without any changes made.  
**Below are the Logs:**

W (78384) wifi:Haven't to connect to a suitable AP now!
E (78394) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (81734) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (81744) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (81754) wifi:Haven't to connect to a suitable AP now!
I (81754) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (81764) chip[DL]: Next WiFi station reconnect in 100 ms
I (81764) chip[DL]: Done driving station state, nothing else to do...
W (81774) wifi:Haven't to connect to a suitable AP now!
E (81774) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (81794) wifi:Haven't to connect to a suitable AP now!
E (81794) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (81864) wifi:Haven't to connect to a suitable AP now!
I (81864) chip[DL]: Attempting to connect WiFi station interface
I (81864) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (81884) chip[DL]: Done driving station state, nothing else to do...
W (81884) wifi:Haven't to connect to a suitable AP now!
E (81894) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (81974) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (81984) wifi:state: init -> auth (b0)
I (81984) wifi:state: auth -> assoc (0)
I (81994) wifi:state: assoc -> run (10)
I (85394) wifi:state: run -> init (fc0)
I (85394) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (85394) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (85394) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (85404) wifi:Haven't to connect to a suitable AP now!
I (85414) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (85424) chip[DL]: Next WiFi station reconnect in 100 ms
I (85434) chip[DL]: Done driving station state, nothing else to do...
W (85434) wifi:Haven't to connect to a suitable AP now!
E (85444) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (85444) wifi:Haven't to connect to a suitable AP now!
E (85464) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (85534) wifi:Haven't to connect to a suitable AP now!
I (85534) chip[DL]: Attempting to connect WiFi station interface
I (85534) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (85544) chip[DL]: Done driving station state, nothing else to do...
W (85554) wifi:Haven't to connect to a suitable AP now!
E (85564) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F

aashikareliya, 2023-07-31:
> @aashikareliya What are you actually trying to do? What changes have you made in the standard examples? Also, are there any logs you can share? Please provide all the relevant details and then [...]

I am trying to do ble commissioning with other manufacturer device which is matter enabled and support BLE 5.3. I am using Light Bulb example in ESP32-C3 without any changes made.
Below are the Logs:

W (78384) wifi:Haven't to connect to a suitable AP now!
E (78394) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (81734) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (81744) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (81754) wifi:Haven't to connect to a suitable AP now!
I (81754) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (81764) chip[DL]: Next WiFi station reconnect in 100 ms
I (81764) chip[DL]: Done driving station state, nothing else to do...
W (81774) wifi:Haven't to connect to a suitable AP now!
E (81774) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (81794) wifi:Haven't to connect to a suitable AP now!
E (81794) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (81864) wifi:Haven't to connect to a suitable AP now!
I (81864) chip[DL]: Attempting to connect WiFi station interface
I (81864) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (81884) chip[DL]: Done driving station state, nothing else to do...
W (81884) wifi:Haven't to connect to a suitable AP now!
E (81894) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (81974) wifinew<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (81984) wifi:state: init -> auth (b0)
I (81984) wifi:state: auth -> assoc (0)
I (81994) wifi:state: assoc -> run (10)
I (85394) wifi:state: run -> init (fc0)
I (85394) wifi🆕<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (85394) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (85394) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (85404) wifi:Haven't to connect to a suitable AP now!
I (85414) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (85424) chip[DL]: Next WiFi station reconnect in 100 ms
I (85434) chip[DL]: Done driving station state, nothing else to do...
W (85434) wifi:Haven't to connect to a suitable AP now!
E (85444) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (85444) wifi:Haven't to connect to a suitable AP now!
E (85464) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (85534) wifi:Haven't to connect to a suitable AP now!
I (85534) chip[DL]: Attempting to connect WiFi station interface
I (85534) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (85544) chip[DL]: Done driving station state, nothing else to do...
W (85554) wifi:Haven't to connect to a suitable AP now!
E (85564) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F

dhrishi, 2023-07-31:
> I am using Light Bulb example in ESP32-C3 without any changes made.

The logs show that the C3 already has some Wi-Fi credentials and hence is trying to connect to it and fails. You will have to reset to factory the device so that it advertises over BLE.

On ESP32-C3 console enter: `matter device factoryreset`

OR

On ESP32-C3, long press the Boot button until there's a print on the console mentioning reset to factory.

aashikareliya, 2023-07-31:
@dhrishi Okay I wiil try and let you know ,if I get any error/issues.

aashikareliya, 2023-08-01:
> The logs show that the C3 already has some Wi-Fi credentials and hence is trying to connect to it and fails. You will have to reset to factory the device so that it advertises over BLE.
>
> On ESP32-C3 [...]

@dhrishi I tried doing this but it is showing same logs as above. Is there ant method to resolve the issue?

dhrishi, 2023-08-01:
It should not auto connect to the Wi-Fi network if reset to factory is successful. Is SSID and passphrase entered through the menuconfig? Can you try:
```
$ cd /path/to/example
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py erase-flash build flash monitor
```