espressif/esp-matter #547: How to get endpoint 0 id in application ?

aggaddam, 2023-07-28:
I would like to read some of the attributes on endpoint 0. But it seems when a node is created by default with device type, esp matter SDK internally creates a node and adds endpoint 0. 

Is there way how I can extract the endpoint 0 id ?

jonsmirl, 2023-07-28:
Then endpoint ID of endpoint 0 is zero.

aggaddam, 2023-07-28:
@jonsmirl Thanks! I thought endpoint ids are different from the endpoint numbers.

I am trying to read the AdminVendorId value after successful commissioning but it fails to read this attribute on endpoint 0. Am I missing anything here ?

```
uint16_t endpoint_id  = 0;
uint32_t cluster_id   = AdministratorCommissioning::Id;
uint32_t attribute_id = AdministratorCommissioning::Attributes::AdminVendorId::Id;;

node_t      *node      = node::get();
endpoint_t  *endpoint  = endpoint::get(node, endpoint_id);
cluster_t   *cluster   = cluster::get(endpoint, cluster_id);
attribute_t *attribute = attribute::get(cluster, attribute_id);

esp_matter_attr_val_t val = esp_matter_invalid(NULL);
if (ESP_OK == attribute::get_val(attribute, &val))
{
    ESP_LOGI(TAG, "AdminVendorId val: %" PRIu16 "",val.val.u16);
}
```

jonsmirl, 2023-07-28:
You should only be reading these from another device - like a phone.

To set vendor/device ID it is in menuconfig

(Top) → Component config → CHIP Device Layer → Device Identification Options

aggaddam, 2023-07-31:
@jonsmirl I am implementing for the matter device (not the controller).  AdminVendorId will be set by the Administrator controller who opens the commissioning window. I want to read this AdminVendorIdattribute to determine who commissions the device (Like Google, Amazon etc..)  for custom implementation on the device side.

[image]

wqx6, 2023-08-07:
There is an `AttributeAccess` for `AdministratorCommissioning` Cluster https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/administrator-commissioning-server/administrator-commissioning-server.cpp#L46, You can't get the right attribute values on clusters with an `AttributeAccess` by `attribute::get_val()`. I suggest you to get these attribute values by `{ClusterName}::Attributes::{AttributeName}::Get()` in https://github.com/project-chip/connectedhomeip/blob/master/zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.h

dhrishi, 2023-08-14:
@aggaddam Please close this if your issue is addressed

dhrishi, 2023-08-27:
Closing as this is addressed. Please re-open if you still face issues

closed 2023-08-27 by dhrishi