espressif/esp-matter #548: Can't link MbedTLS after last commit

Okywan, 2023-07-30:
After last commit, in which refrigerator and room_air_conditioner were added, I cannot compile anymore my project because the compile cannot find MbedTLS:

`../esp-idf/tools/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:330: undefined reference to `mbedtls_hkdf'`

Tried updating esp-idf to 5.1 version and freshly installed esp-matter again, all with the same result.

```
 esp-idf % git log
commit cbce221e88d52665523093b2b6dd0ebe3f1243f1 (HEAD, tag: v5.1)
Merge: c570f67461 e44a6888ea
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Wed Jun 28 17:33:53 2023 +0800

    Merge branch 'bugfix/mac_ext_order_for_c6_h2_v5.1' into 'release/v5.1'
    
    efuse: Fix byte order in MAC_EXT (for C6 and H2) (v5.1)
    
    See merge request espressif/esp-idf!23990
```

```
git log
commit 85963d2ca8f4ecb49fa3234f5b0273cb2df411b7 (HEAD -> main, origin/main, origin/HEAD)
Merge: 714233a de530ef
Author: Shu Chen <chenshu@espressif.com>
Date:   Thu Jul 27 20:22:48 2023 +0800

    Merge branch 'nvs/change_default_nvs_size_to_48k' into 'main'
    
    change example default nvs size to 48k
    
    See merge request app-frameworks/esp-matter!443
```

sayondeep, 2023-07-30:
@Okywan You need to enable HKDF algorithm by setting MBEDTLS_HKDF_C = y in sdkconfig.
 You can also do this by running `idf.py menuconfig` from your project directory. You can find HKDF algorithm under component config --> mbedTLS

Okywan, 2023-07-31:
Right, that was it. Is this explained somewhere and did I miss it?

closed 2023-07-31 by Okywan