espressif/esp-matter #549: Increase the dynamic endpoints

HARRYhelenH, 2023-07-31:
Hello,
I'm currently working on bridge-app examples to build bridge between matter and non-matter devices.
In the examples bridge-app/esp32 the max endpoints can be added is 16. Unable to find the macro configuration in the menuconfig.

Tried including the file <config/standalone/CHIPProjectConfig.h> and changing the macro value, but it is overridden as 16 endpoints.

jonsmirl, 2023-07-31:
CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT
CONFIG_ESP_MATTER_AGGREGATOR_ENDPOINT_COUNT

HARRYhelenH, 2023-08-01:
could you please tell the where the macros are defiend. unable to find macro definiton.
esp-idf:v4.4.3
matter:release/v1.0

sayondeep, 2023-08-01:
@HARRYhelenH you can change the setting under Component config -> ESP Matter -> Maximum device types per endpoint by running `idf.py menuconfig`.

HARRYhelenH, 2023-08-01:
In the bridge-app/esp32 examples I don't have this configuration.
But the endpoints can be increased in the cmakelist file  by changing the count in the command:
**idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++14;-Os;-DCHIP_HAVE_CONFIG_H;-DCHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT=16" APPEND)**

closed 2023-08-01 by HARRYhelenH