espressif/esp-matter #550: ESP32 C6 DevKitC 1: chip-tool command pairing ble-wifi not working

manamohan1, 2023-08-01:
Hi,
    I did the esp-matter build for ESP32 C6 Devkit for Light control example. I successfully did the build and flashed the image. 
On the Linux PC I am able to start chip-tool in interactive session but when I execute  "pairing ble-wifi 42 manu password 20202021 3840"
I am getting the following error
================================
>>> pairing ble-wifi 42 manu password 20202021 3840
[1690888667.281245][57489:57489] CHIP:TOO: Command: pairing ble-wifi 42 manu password 20202021 3840 
[1690888667.281570][57489:57491] CHIP:CTL: Setting wifi credentials from parameters
[1690888667.281586][57489:57491] CHIP:CTL: Setting attestation nonce to random value
[1690888667.281631][57489:57491] CHIP:CTL: Setting CSR nonce to random value
[1690888667.281646][57489:57491] CHIP:CTL: Commission called for node ID 0x000000000000002A
[1690888667.285099][57489:57490] CHIP:BLE: BLE removing known devices.
[1690888667.285802][57489:57490] CHIP:BLE: BLE initiating scan.
[1690888667.412437][57489:57491] CHIP:DL: Long dispatch time: 131 ms, for event type 2
[1690888667.532926][57489:57490] CHIP:BLE: Device 48:7E:48:2F:25:CC does not look like a CHIP device.
[1690888667.533659][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888667.534374][57489:57490] CHIP:BLE: Device 20:EE:B0:88:B2:C8 does not look like a CHIP device.
[1690888667.571931][57489:57490] CHIP:BLE: Device 30:A3:C0:B8:30:C7 does not look like a CHIP device.
[1690888667.622910][57489:57490] CHIP:BLE: Device 44:1C:A8:42:A6:10 does not look like a CHIP device.
[1690888667.653712][57489:57490] CHIP:BLE: Device 40:4D:AC:D6:4D:35 does not look like a CHIP device.
[1690888667.812815][57489:57490] CHIP:BLE: Device 07:5D:57:B6:FC:64 does not look like a CHIP device.
[1690888667.863317][57489:57490] CHIP:BLE: Device 44:2D:25:C9:F4:93 does not look like a CHIP device.
[1690888668.311349][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888668.312427][57489:57490] CHIP:BLE: Device FD:29:10:43:E6:F4 does not look like a CHIP device.
[1690888668.523225][57489:57490] CHIP:BLE: Device 7B:B9:CC:8C:07:11 does not look like a CHIP device.
[1690888668.573773][57489:57490] CHIP:BLE: Device 47:DD:AA:C2:8A:13 does not look like a CHIP device.
[1690888669.202486][57489:57490] CHIP:BLE: Device 80:8A:BD:F2:E1:3B does not look like a CHIP device.
[1690888669.703676][57489:57490] CHIP:BLE: Device 7F:16:DB:CA:5C:E8 does not look like a CHIP device.
[1690888669.966699][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888669.991309][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888669.992106][57489:57490] CHIP:BLE: Device C6:79:D8:94:BA:3D does not look like a CHIP device.
[1690888670.203950][57489:57490] CHIP:BLE: Device 4D:69:CA:84:9E:08 does not look like a CHIP device.
[1690888670.336013][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888670.380784][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888670.461089][57489:57490] CHIP:BLE: Device 49:CC:7D:53:F6:16 does not look like a CHIP device.
[1690888670.721709][57489:57490] CHIP:BLE: Device 6A:C3:3F:52:FE:3F does not look like a CHIP device.
[1690888670.833551][57489:57490] CHIP:BLE: Device 56:D0:9E:C4:68:DC does not look like a CHIP device.
[1690888670.834298][57489:57490] CHIP:BLE: Device 10:2B:41:44:F1:2D does not look like a CHIP device.
[1690888670.931546][57489:57490] CHIP:BLE: Device FB:8F:F0:24:2A:50 does not look like a CHIP device.
[1690888671.098345][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888671.098471][57489:57490] CHIP:BLE: Device 48:7E:48:2F:25:CC does not look like a CHIP device.
[1690888671.140642][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888671.172032][57489:57490] CHIP:BLE: Device 44:2D:25:C9:F4:93 does not look like a CHIP device.
[1690888671.172879][57489:57490] CHIP:BLE: Device D7:AB:77:E6:36:14 does not look like a CHIP device.
[1690888671.221590][57489:57490] CHIP:BLE: Device 44:1C:A8:42:A6:10 does not look like a CHIP device.
[1690888671.302832][57489:57490] CHIP:BLE: Device 4D:69:CA:84:9E:08 does not look like a CHIP device.
[1690888671.983095][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888671.983948][57489:57490] CHIP:BLE: Device 44:2D:25:C9:F4:93 does not look like a CHIP device.
[1690888671.984813][57489:57490] CHIP:BLE: Device E5:DB:84:7E:8C:23 does not look like a CHIP device.
[1690888672.011942][57489:57490] CHIP:BLE: Device 07:5D:57:B6:FC:64 does not look like a CHIP device.
[1690888672.401967][57489:57490] CHIP:BLE: Device 7B:B9:CC:8C:07:11 does not look like a CHIP device.
[1690888672.482206][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888672.644262][57489:57490] CHIP:BLE: Device 58:15:D3:DB:6C:3B does not look like a CHIP device.
[1690888672.644389][57489:57490] CHIP:BLE: Device 7B:B9:CC:8C:07:11 does not look like a CHIP device.
[1690888673.141970][57489:57490] CHIP:BLE: Device 47:DD:AA:C2:8A:13 does not look like a CHIP device.
[1690888673.246414][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888673.247334][57489:57490] CHIP:BLE: Device 47:DD:AA:C2:8A:13 does not look like a CHIP device.
[1690888673.247451][57489:57490] CHIP:BLE: Device 80:8A:BD:F2:E1:3B does not look like a CHIP device.
[1690888673.275447][57489:57490] CHIP:BLE: Device 48:7E:48:2F:25:CC does not look like a CHIP device.
[1690888673.563216][57489:57490] CHIP:BLE: Device 7F:16:DB:CA:5C:E8 does not look like a CHIP device.
[1690888673.774209][57489:57490] CHIP:BLE: Device 49:CC:7D:53:F6:16 does not look like a CHIP device.
[1690888673.774467][57489:57490] CHIP:BLE: Device 58:15:D3:DB:6C:3B does not look like a CHIP device.
[1690888674.193313][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888674.193818][57489:57490] CHIP:BLE: Device 44:2D:25:C9:F4:93 does not look like a CHIP device.
[1690888674.323294][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888674.323418][57489:57490] CHIP:BLE: Device FD:29:10:43:E6:F4 does not look like a CHIP device.
[1690888674.530736][57489:57490] CHIP:BLE: Device 4D:69:CA:84:9E:08 does not look like a CHIP device.
[1690888674.585704][57489:57490] CHIP:BLE: Device 40:4D:AC:D6:4D:35 does not look like a CHIP device.
[1690888674.586280][57489:57490] CHIP:BLE: Device 56:D0:9E:C4:68:DC does not look like a CHIP device.
[1690888674.662184][57489:57490] CHIP:BLE: Device 7B:B9:CC:8C:07:11 does not look like a CHIP device.
[... 7 lines cut ...]
[1690888675.163340][57489:57490] CHIP:BLE: Device 47:DD:AA:C2:8A:13 does not look like a CHIP device.
[1690888675.631931][57489:57490] CHIP:BLE: Device 10:2B:41:44:F1:2D does not look like a CHIP device.
[1690888675.714210][57489:57490] CHIP:BLE: Device D0:6E:13:D1:5E:C1 does not look like a CHIP device.
[1690888675.714366][57489:57490] CHIP:BLE: Device 20:EE:B0:88:B2:C8 does not look like a CHIP device.
[1690888675.871040][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888675.872423][57489:57490] CHIP:BLE: Device F8:AB:0F:0F:85:55 does not look like a CHIP device.
[1690888676.052739][57489:57490] CHIP:BLE: Device 44:1C:A8:42:A6:10 does not look like a CHIP device.
[1690888676.422021][57489:57490] CHIP:BLE: Device 40:4D:AC:D6:4D:35 does not look like a CHIP device.
[1690888676.505286][57489:57490] CHIP:BLE: Device 48:7E:48:2F:25:CC does not look like a CHIP device.
[1690888676.505684][57489:57490] CHIP:BLE: Device 45:12:9C:B3:32:D6 does not look like a CHIP device.
[1690888676.506353][57489:57490] CHIP:BLE: Device 80:8A:BD:F2:E1:3B does not look like a CHIP device.
[1690888676.634821][57489:57490] CHIP:BLE: Device CE:B1:01:E7:E6:2C does not look like a CHIP device.
[1690888676.680810][57489:57490] CHIP:BLE: Device 48:7E:48:2F:25:CC does not look like a CHIP device.
[1690888676.892032][57489:57490] CHIP:BLE: Device 49:CC:7D:53:F6:16 does not look like a CHIP device.
[1690888676.975482][57489:57490] CHIP:BLE: Device 4B:7C:48:07:79:0C does not look like a CHIP device.
[1690888676.976278][57489:57490] CHIP:BLE: Device 49:CC:7D:53:F6:16 does not look like a CHIP device.
[... cut ...]
[1690888677.422661][57489:57491] CHIP:TOO: Pairing Failure: src/platform/Linux/bluez/ChipDeviceScanner.cpp:175: CHIP Error 0x00000032: Timeout
[1690888677.422674][57489:57491] CHIP:TOO: Run command failure: src/platform/Linux/bluez/ChipDeviceScanner.cpp:175: CHIP Error 0x00000032: Timeout
[1690888677.422695][57489:57491] CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:175: CHIP Error 0x00000032: Timeout
[1690888677.425138][57489:57491] CHIP:BLE: Scan complete. No matching device found.
===========================================================================

shubhamdp, 2023-08-02:
> [1690888677.425138][57489:57491] CHIP:BLE: Scan complete. No matching device found.

Can you please reset the C6 and give another try. Also, would appreciate the logs from C6 (please attach then in a file rather than pasting). Also, share the esp-matter and esp-idf commits IDs.

manamohan1, 2023-08-04:
I tried resetting C6 and it is not working.
As requested please find the attached chip-tool log and C6 console log files.
[c6.log](https://github.com/espressif/esp-matter/files/12256798/c6.log)
[chip-tool.log](https://github.com/espressif/esp-matter/files/12256799/chip-tool.log)

manamohan1, 2023-08-04:
~/ESP/esp-matter$ git rev-parse --short HEAD
85963d2
==================================================
~/ESP/esp-idf$ git rev-parse --short HEAD
ea5e0ff298
==============================================

manamohan1, 2023-08-05:
Any input on this issue ?, please help.

dhrishi, 2023-08-06:
@manamohan1 The problem here is that the ESP32 is unable to connect to the Wi-Fi network during the commissioning process.

```
I (153639) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (153639) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (153649) wifi:Haven't to connect to a suitable AP now!
I (153649) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (153659) chip[DL]: Done driving station state, nothing else to do...
W (153659) wifi:Haven't to connect to a suitable AP now!
E (153669) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (153679) wifi:Haven't to connect to a suitable AP now!
```

Can you please confirm if:
1. The AP being connected to is 2.4 GHz and not 5 GHz
2. It is within the range for the device to connect to

If yes, then first please try out `esp-idf/examples/wifi/getting_started/station/` to connect to the same AP and see how it works.

dhrishi, 2023-08-10:
@manamohan1 Please close this if your issue is resolved

dhrishi, 2023-08-27:
Closing this now

closed 2023-08-27 by dhrishi

manamohan1, 2023-08-29:
It is resolved