espressif/esp-matter #551: Wifi control with ESP-Matter

smeisner, 2023-08-02:
I am working on an ESP32-S3 based device that has a touch screen and can use it to provide wifi credentials (SSID and PSK). I do not want to rely on commissioning via bluetooth. Allowing the user to use the touchscreen is way easier!

If I start wifi the usual way (`WiFi.begin(ssid, pass)`) the credentials get cached. Then on the next reboot, calling `esp_matter::start()` uses this info to connect to my WLAN. Of course, if I also call `WiFi.begin()`, that crashes.

This device will serve a web page (for config), use SNTP for time and a few other uses for wifi. So my questions;
1. Can I call `WiFi.begin(ssid, pass)` and provide the instance to Matter?
2. Is there an API to manipulate the instance of wifi being used by Matter? For example, if the connection drops, I want to re-establish the WLAN connection ... or read the signal strength to report to the user via the GUI. Or change the wifi setup (SSID) locally and re-establish the wifi connection.

shubhamdp, 2023-08-02:
@smeisner Are you using Arduino library?

1. Yes, you can connect the device to wifi and start the matter, there is no need to reboot the device.
2. Matter's wifi manager will re-establish connection. For reading signal strength and similar stuff, you can use the APIs from [esp_wifi.h](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html).

smeisner, 2023-08-02:
Thanks for the reply, but this is not what I am seeing. If I start wifi then call to start Matter, I get:
```
[19571.00h.28m.37s.119] [  2099][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY 
[19571.00h.28m.37s.176] [  2140][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started 
[19571.00h.28m.37s.176] [  2140][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0 
[19571.00h.28m.37s.176] [  2140][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START 
[19571.00h.28m.37s.278] [  2248][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: xxxxxx, BSSID: e6:63:da:6b:ea:78, Channel: 3, Auth: WPA2_PSK 
[19571.00h.28m.37s.278] [  2249][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED 
[19571.00h.28m.37s.787] [  2751][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.0.100 
[19571.00h.28m.37s.787] [  2751][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP 
[19571.00h.28m.37s.787] [  2754][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.0.100, MASK: 255.255.255.0, GW: 192.168.0.254 
[19571.00h.28m.38s.666] .Ready 
[19571.00h.28m.38s.666] Host name: thermostat 
[19571.00h.28m.38s.666] IP address: 192.168.0.100 
[19571.00h.28m.41s.866] Starting Matter API 
[19571.00h.28m.41s.925] [  6873][I][matter.cpp:121] MatterInit(): [Thermostat] Thermostat endpoint_id = 1 
[19571.00h.28m.41s.925] [  6873][E][matter.cpp:132] MatterInit(): [Thermostat] Thermostat heating cluster id = 1 
[19571.00h.28m.41s.925] [  6881][E][matter.cpp:145] MatterInit(): [Thermostat] Thermostat cooling cluster id = 2 
[19571.00h.28m.41s.925] [  6889][E][matter.cpp:177] MatterInit(): [Thermostat] Calling esp_matter::start() 
[19571.00h.28m.41s.925]  
[19571.00h.28m.41s.925] E (6450) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3fcebe44 (config or if_key is NULL or duplicate key) 
[19571.00h.28m.41s.925] 
[19571.00h.28m.41s.957] assert failed: esp_netif_create_default_wifi_ap wifi_default.c:319 (netif) 
[19571.00h.28m.41s.994] 
```

smeisner, 2023-08-03:
Thanks ... I haven't tried this yet, but when you say " ... and perform connect" are you referring to wifi connect? If so, that would be calling `WiFi.begin()` which does init plus connect.

After walking thru quite a bit of CHIP/Matter code (...and ESP-matter code ... and Arduino ESP-matter code), it does seem the problem is both `WiFi.begin` and `esp_matter::start()` are trying to both initialize the wifi stack. Neither the Matter API (nor the ESP API) are clean enough to figure out a way around this...at least for me.

I do have things working as far as wifi is up (via starting matter) and I can provide a web server on the interface. But none of the Arduino wifi API calls work (get hostname, IP addr, RSSI, etc). But this is only working since the SSID and PSK are already in NVS. If I clear that, it does not work. If the user needs to select a different SSID (or first time running the device), this solution does not work at all.

smeisner, 2023-08-05:
I have posted a question in the esp32-arduino-matter issues board (https://github.com/Yacubane/esp32-arduino-matter/issues/36) since I am using that code (with the Arduino framework) and cannot open/close the commissioning window.

@shubhamdp In your last reply, you mentioned "(But Matter code should handle if wifi is already initialized, will fix this)." Is this something on the roadmap?

shubhamdp, 2023-08-07:
@smeisner Thanks for trying out the steps. Need to put some guards here: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32Utils.cpp#L319.

I am adding this to the roadmap, will get this fix as soon as possible.

smeisner, 2023-08-12:
I have managed to get things working. Removing all Arduino API calls to wifi and using ESP-IDF calls really helped out. The real problem is coordinating taking the stack down before switching to Matter ... or vice versa. 

@shubhamdp For the most part, I used the suggestions you provided. The problem I encountered was the ESP-Arduino-Matter library doesn't have full support, so I was unable to open/close the provisioning window.

Code has been checked into the branch named 'Matter' in my github repo: https://github.com/smeisner/smart-thermostat. The interesting modules are https://github.com/smeisner/smart-thermostat/blob/matter/app/src/matter.cpp and https://github.com/smeisner/smart-thermostat/blob/matter/app/src/wifi.cpp

Keep in mind this project is still a work in progress. So don't beat me up on code quality ...yet!!  :D

closed 2023-08-12 by smeisner

smeisner, 2023-08-14:
BTW, I still believe allowing Matter to start after the wifi stack is initialized is still very valuable. The work I did in my repo is not trivial and I feel it is possibly going to crash due to the complexity. So I am still looking forward to the feature you mentioned in https://github.com/espressif/esp-matter/issues/551#issuecomment-1667825553

shubhamdp, 2023-09-01:
@smeisner please take a look at https://github.com/project-chip/connectedhomeip/pull/29017.

smeisner, 2023-09-01:
@shubhamdp Thanks for the heads-up!

Line 336 is exactly what I need, but the entire change set looks great! When this is merged, I will certainly give the new code a try.

smeisner, 2023-09-01:
@shubhamdp I just had a thought; in the current implementation, if wifi connection drops, matter will initiate a reconnect. With the change in this PR, who will be responsible for re-connecting to the AP? I am hoping/assuming not matter.

shubhamdp, 2023-09-04:
@smeisner Even in this case, reconnection will be managed by chip stack. https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32Utils.cpp#L365.

> I am hoping/assuming not matter.

There is a big state machine here, which would need some though for making it happen.

smeisner, 2023-09-04:
Ok good to know. I would have assumed incorrectly!