espressif/esp-matter #552: Lost ability to commission at Google

jonsmirl, 2023-08-02:
I can't commission anymore using the Google Sample app. However, I can still commission on Google Home.

But if I commission on Google Home and share to the sample app, I get this error again on the share.
https://issuetracker.google.com/issues/293901764
[sdkconfig.txt](https://github.com/espressif/esp-matter/files/12235378/sdkconfig.txt)

Sometimes I am getting error kDacVendorIdMismatch

~~~
2023-08-01 22:17:15.834 11954-11954 SetupDeviceViewModel    com.google.android.gms.ui            E  Commissioning failed with state Attestation failure. [CONTEXT service_id=336 ]
                                                                                                    m.etd: Device did not pass attestation (error(s)=[1, 5, 11])
                                                                                                    	at m.eui.g(:com.google.android.gms.optional_home@232617065@23.26.17 (100400-0):509)
                                                                                                    	at m.euc.b(:com.google.android.gms.optional_home@232617065@23.26.17 (100400-0):13)
                                                                                                    	at m.pko.f(:com.google.android.gms.optional_home@232617065@23.26.17 (100400-0):14)
                                                                                                    	at m.pst.run(:com.google.android.gms.optional_home@232617065@23.26.17 (100400-0):114)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8757)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
~~~
~~~
./mfg_tool.py -cn "Test" -v 0xFFF2 -p 0x8001 --pai \
-k ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd ../../connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Lowpan" --product-name "Dimmer" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840 \
--fixed-labels "1/name/Main" "2/name/BTN-2" "3/name/BTN-3" "4/name/BTN-4" "5/name/BTN-5" "6/name/BTN-6" "7/name/BTN-7"  "8/name/BTN-8" \
--supported-modes "0/Switch/1/1\1" "1/Dimmer/1/1\1" "2/Smartbulb/1/1\1"
~~~

jonsmirl, 2023-08-03:
This magically started working again today.

jonsmirl, 2023-08-03:
And now it is is not working again.

dhrishi, 2023-08-27:
I assume this is fixed on the app side. I could commission the device using the Google Home Sample App, a couple of days back.

closed 2023-08-27 by dhrishi

jonsmirl, 2023-08-27:
I think Google is breaking this on their servers.