espressif/esp-matter #553: Supply supported modes for Mode Select cluster from application code

itavero, 2023-08-02:
In the [documentation related to Mode Select](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#mode-select), it is mentioned that the mfg_tool should be used to configure the supported modes.
First of all, I find the documentation a bit vague, but most importantly, I don't think this is a good solution for our product.

In our product, the supported modes will depend on the capabilities of the appliance it is connected to. Secondly, the product is multi-lingual, and it is likely that the labels for the different modes will have to be multi-lingual as well.
This seems cumbersome to manage when this data is stored in flash somewhere, so I would prefer to provide this information from the application code.

Is it possible to provide the entire configuration for the Mode Select Cluster from the application code?
If so, where can I find a reference for this?

jonsmirl, 2023-08-02:
1) The mode select data is stored in flash and you can use the ESP NVS API to write it. But that probably isn't what you want. That data is supposed to be fixed.
2) Probably what you are looking for is the UserLabel cluster. So when you change language you can alter the UserLabel cluster which is a normal CHIP operation.  The UI is going to have to look for these overrides and display them if present. You can also use this to allow the user to edit the names.
3) Mode select is changing a lot in CHIP 1.2. I don't have access to see how it is being changed. Maybe the Espressif people will tell us. For example I know Dishwashers are coming and they use the new Mode Select.
4) Multilingual support is not working in current CHIP. Doing things like setting the code page has no impact. But you can write the country code in, and then read it back in your app and then select different things in the UserData cluster.
5) None of Google, Apple, Amazon support Mode Select.  You have to use your own app.

If you are doing this in your own code you can pretty much ignore all of the strings in Matter and just work off from the mode numbers. If the international UI is handled in your app -- just ignore UserLabels except for personalized labels.

itavero, 2023-08-03:
Interesting. I looked at the PICS for the software components used by Apple and Google and they mention Mode Select cluster compatibility, but I guess they do not expose it in their default apps then. That's too bad (one of our goals with this Matter integration is to prevent all our OEM customers from having to create/maintain their own app for controlling their systems).

jonsmirl, 2023-08-03:
Apple/Google/Amazon are going to do it. But last I checked they have not doing it.,

dhrishi, 2024-10-15:
@itavero Closing this now. You can try out the API and update in case of any issues

closed 2024-10-15 by dhrishi