espressif/esp-matter #554: IOS 17 problem with Generic Switch

Kristian8606, 2023-08-03:
I have a problem with IOS 17 and Generic Switch, after I add 2 buttons they have the same name
button 1, button 1.
if i add 4 buttons, homekit shows them as button 1, button 1, button 1, button 3. gpio_6, gpio_16 and gpio_17, react to button 3 and gpio_0 reacts to the first button 1.
Can anyone confirm if this is an ios 17 issue or a generic_switch example not work?

XAT-Project, 2023-08-09:
Hi @Kristian8606,

I have tested the generic switch example from the main branch with the following configuration:

| Device       | Firmware update |
| ------------ | --------------- |
| Homepod mini | 16.6            |
| iPhone 12    | 16.6            |
| ESP32S3      | /               |

All worked well for me with this configuration when I setup 4 buttons, however all of this are called "Button1" in my case.

Can you please tell me what ESP you are using, your controller (Homepod, Apple TV, etc.), and the version of your controller?

In addition, could you share a snippet of how you set up your button?

Kristian8606, 2023-08-09:
@XAT-Project  Thanks for responding.
 Apparently I hadn't figured out exactly how to add buttons, but now, i built the example from the master branch and added 4 buttons, they work correctly but they are all called Button1.
Now the question is why all the buttons are called Button1 and what should be done to call them Button1, Button2, Button3, Button4?
Sorry for my bad english.

shripad621git, 2023-08-09:
@Kristian8606 I have tried out the **generic switch** example by adding buttons to check for whether there is any bug in the code .It does not seem to be the case .The mapping of gpios to buttons seems to be unique. However with apple home all the buttons appear as Button1  which was the case previously as well .So ,it seems to be an ios issue.

Kristian8606, 2023-08-09:
I have an Aqara H1 2-Gang wireless switch with HUB M1S running Matter Beta where the buttons are correctly spelled Button1 and Button2, this makes me think the problem is not HomeKit.

jvmahon, 2024-01-23:
> I have an Aqara H1 2-Gang wireless switch with HUB M1S running Matter Beta where the buttons are correctly spelled Button1 and Button2, this makes me think the problem is not HomeKit.

@Kristian8606 Were you able to get the button labels to work on espressif?  If so, can you share the code that solved how to label the buttons and have iOS Home recognize it.  Does it also work in iOS 17?

Thank you!

Kristian8606, 2024-01-23:
It still doesn't display the buttons properly