espressif/esp-matter #555: CHIP_SYSTEM_CONFIG_PACKETBUFFER_POOL_SIZE

jonsmirl, 2023-08-03:
Can Espressif please check in their platform code and see if this is being properly dealt with?  CHIP_SYSTEM_CONFIG_PACKETBUFFER_POOL_SIZE
[buffer.txt](https://github.com/espressif/esp-matter/files/12252680/buffer.txt)
Looking at this log, I booted and ran out of packet buffers in 30 seconds, but I still had 21KB of free memory. What is new is that this device has two saved bindings and when it boots it is setting up sessions to the bound devices. Setting up those sessions is using packet buffers.

Note that K32W0 platform and CYW30739 platform both set this number to zero.  Silabs and QPG are setting it to 8. The ESP code does not touch it so I think it is defaulting to 15?  Is the ESP platform duplicating these buffers? Once in CHIP and once in the IDF?

Just increasing the number of buffer may not be the fix here, it might be better to block the buffer requests until some things complete.

jonsmirl, 2023-08-04:
This log shows how I can boot and immediately run out of packet buffers.
[buffer.txt](https://github.com/espressif/esp-matter/files/12261530/buffer.txt)

wqx6, 2023-08-07:
In ESP-IDF, the `MEMP_MEM_MALLOC` is enabled, which means that we will use `mem_malloc(malloc)`(heap) instead of LwIP mem pool to allocate memory for LwIP pbuf. That's the reason why the `CHIP_SYSTEM_CONFIG_PACKETBUFFER_POOL_SIZE` affects nothing. So when there are some errors on `pbuf_alloc()`, the heap should be used up. Can you please print the heap memory on the time when `pbuf_alloc()` fails instead in `app_main()`? That would be a more accurate estimation for the heap.

jonsmirl, 2023-09-03:
@wqx6 finally caught it with heap dump code inserted. It is not out of memory.

~~~
I (550900) NimBLE: GAP procedure initiated: stop advertising.

E (557860) chip[CSL]: PacketBuffer: pool EMPTY. B
Heap summary for capabilities 0x00000800:
  At 0x3fcab270 len 201888 free 8872 allocated 188340 min_free 1504
    largest_free_block 1472 alloc_blocks 934 free_blocks 9 total_blocks 943
  At 0x3fcdc710 len 10576 free 3344 allocated 6800 min_free 1808
    largest_free_block 1408 alloc_blocks 4 free_blocks 3 total_blocks 7
  At 0x50000010 len 8152 free 2828 allocated 4932 min_free 2660
    largest_free_block 1408 alloc_blocks 3 free_blocks 2 total_blocks 5
  Totals:
    free 16040 allocated 199092 min_free 5972 largest_free_block 1472
I (557860) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (557860) chip[DIS]: Failed to reply to query: b
E (557860) chip[CSL]: PacketBuffer: pool EMPTY. B
Heap summary for capabilities 0x00000800:
  At 0x3fcab270 len 201888 free 9868 allocated 187360 min_free 1504
    largest_free_block 1472 alloc_blocks 930 free_blocks 12 total_blocks 942
  At 0x3fcdc710 len 10576 free 3344 allocated 6800 min_free 1808
    largest_free_block 1408 alloc_blocks 4 free_blocks 3 total_blocks 7
  At 0x50000010 len 8152 free 4364 allocated 3400 min_free 2660
    largest_free_block 1408 alloc_blocks 2 free_blocks 3 total_blocks 5
  Totals:
    free 17576 allocated 197560 min_free 5972 largest_free_block 1472
E (557870) chip[DIS]: Failed to reply to query: b

~~~

jonsmirl, 2023-09-06:
sdkconfig  
[sdkconfig.txt](https://github.com/espressif/esp-matter/files/12537634/sdkconfig.txt)

wqx6, 2023-09-07:
Looks like the largest free block is too small for a pbuf memory allocating.

> How can I reduce RAM consumption?

May you refer to this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html?highlight=heap#minimizing-ram-usage

jonsmirl, 2023-09-18:
The esp-matter stack is getting too big for the C3 if you do anything more than the most basic function -- ie light bulb.

The problem is dynamic RAM consumption, not static. There are too many tiny allocations with their associated overhead.  I suspect the best way to address this is to allocate the memory for clusters as a single allocation instead of dozens of tiny allocations. Meanwhile I have moved on the the S3 with 2MB of PSRAM. I'm only using 75KB of that 2MB, but since there is lots of RAM now my allocations never fail.

jonsmirl, 2023-10-31:
This is a symptom of the slow leak problem  https://github.com/espressif/esp-matter/issues/674

There also appear to be a bug in CHIP minimal DNS https://github.com/project-chip/connectedhomeip/issues/29800

closed 2023-10-31 by jonsmirl