espressif/esp-matter #556: chip-tool commissioning error: Failed to verify peer's MAC

albrownwood, 2023-08-05:
The device is already commissioned and part of an Apple fabric, my understanding is that I can pair this same device using chip-tool so it becomes part of a second fabric. When doing so I get the message "Failed to verify peer's MAC" in chip-tool console.

[chip_tool_commissioning_issue.txt](https://github.com/espressif/esp-matter/files/12265326/chip_tool_commissioning_issue.txt)

dhrishi, 2023-08-05:
@albrownwood Did you open the commissioning window using Apple Home? If yes, it would have generated a new QR code/manual code (enhanced commissioning). You can then use the following command from chip-tool:

```
chip-tool pairing code <new-node-id> <code-from-Apple-Home>
```

albrownwood, 2023-08-05:
@dhrishi yes the commissioning window was opened with Apple home, pairing was successful with your indications, thanks!

closed 2023-08-05 by albrownwood

gautam1joshi, 2023-08-07:
hello @albrownwood ,
which release of esp-IDF are you using because i get error when i try to commission the device over wifi?
your help will be appreciated.