espressif/esp-matter #557: Update documentation to fix two issues with WSL

Kampi, 2023-08-05:
Please reference the [Prerequisites](https://github.com/espressif/connectedhomeip/blob/v1.0.0.2/docs/guides/BUILDING.md#installing-prerequisites-on-linux) when using WSL and Ubuntu. I wasn´t able to run `install.sh` until I had installed the missing packages.

Also, I needed to run `install.sh` with `sudo` because of a permission issue.

sayondeep, 2023-08-12:
@Kampi Thanks for the feedback.
`install.sh` should not required sudo privileges if the repo was cloned from within the WSL itself.
The requirement of sudo may arise if the repo was cloned in windows and accessed from inside WSL.
OR It was cloned as root in WSL and accessed by other user on WSL.
OR The installation directory of WSL was changed using `wsl --export`.

dhrishi, 2023-09-04:
Closing this now

closed 2023-09-04 by dhrishi