espressif/esp-matter #558: New Modeselect without Semantic Tags

jonsmirl, 2023-08-06:
When I read the ModeSelect cluster with wildcard subscription, TLV encoder is dying. This was working before semantic tag was removed.
~~~
./mfg_tool.py -cn "Test" -v 0xFFF2 -p 0x8001 --pai \
-k ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd ../../connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Lowpan" --product-name "Dimmer" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840 \
--fixed-labels "1/name/Main" "2/name/BTN-2" "3/name/BTN-3" "4/name/BTN-4" "5/name/BTN-5" "6/name/BTN-6" "7/name/BTN-7"  "8/name/BTN-8" \
--supported-modes "0/Switch/1" "1/Dimmer/1" "2/Smartbulb/1"
~~~
~~~
I (116103) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x0000FFFC is 0 **********
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x400588a2  RA      : 0x4208b0e4  SP      : 0x3fcc1610  GP      : 0x3fc95000  
Stack dump detected
0x400588a2: _data_end_btdm_rom in ROM

0x4208b0e4: chip::TLV::TLVWriter::WriteData(unsigned char const*, unsigned long) at /home/jonsmirl/lowpanC3/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:786
 (inlined by) chip::TLV::TLVWriter::WriteData(unsigned char const*, unsigned long) at /home/jonsmirl/lowpanC3/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:760

TP      : 0x3fc7a4b0  T0      : 0x00000132  T1      : 0x00000389  T2      : 0xffffffff  
S0/FP   : 0x3fcc1870  S1      : 0x0000000c  A0      : 0x3fcd63b1  A1      : 0x00000000  
A2      : 0x0000000c  A3      : 0x0000000c  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0xfffffffa  S2      : 0x0000000c  S3      : 0x00000000  
S4      : 0x3fcc1884  S5      : 0x3fcc187c  S6      : 0x0000ffff  S7      : 0x00000000  
S8      : 0x00000002  S9      : 0x0000ffff  S10     : 0x3c14d000  S11     : 0x42022000  
0x42022000: emberAfLongStringLength(unsigned char const*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include/nlio-base.h:151
 (inlined by) GetUnaligned16 at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include/nlio-base.hpp:145
 (inlined by) GetUnaligned16 at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include/nlio-byteorder-little.hpp:653
 (inlined by) chip::Encoding::LittleEndian::Get16(unsigned char const*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/CHIPEncoding.h:312
 (inlined by) emberAfLongStringLength(unsigned char const*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/message.cpp:35

T3      : 0x0000005a  T4      : 0x3fcd63ad  T5      : 0x3fcd6024  T6      : 0x3fcc189c  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x400588a2 in ?? ()
#0  0x400588a2 in ?? ()
#1  0x4208b0e4 in chip::TLV::TLVWriter::WriteData (len=12, p=0x0, this=<optimized out>) at ../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:785
#2  chip::TLV::TLVWriter::WriteData (this=0x3fcc1870, p=<optimized out>, len=<optimized out>) at ../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:760
#3  0xfffffff9 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
ELF file SHA256: 9e2b9fcdf5f481b9

~~~

jonsmirl, 2023-08-06:
https://github.com/project-chip/connectedhomeip/issues/28544#issuecomment-1666890618

In the ESP Platform code, (examples/platform/esp32/mode-support/static-supported-modes-manager.cpp maybe?) when there are no semantic tags, do you build a zero length array and initialize it?  If it is not initialized you'd see this error.

dhrishi, 2023-08-10:
@jonsmirl With https://github.com/espressif/esp-matter/issues/567 , can this issue be closed or is it still a problem?

closed 2023-08-10 by jonsmirl