espressif/esp-matter #559: crash when pairing ble-wifi using chip-tool: Chip stack locking error

albrownwood, 2023-08-07:
using chip-tool on macOS to pair an esp32-based device gives esp32 crash with: 
"Chip stack locking error at 'src/system/SystemLayerImplFreeRTOS.cpp:101'. Code is unsafe/racy"

[repro_pairing_crash.txt](https://github.com/espressif/esp-matter/files/12270990/repro_pairing_crash.txt)
[esp_crash_pairing_chiptool.txt](https://github.com/espressif/esp-matter/files/12270992/esp_crash_pairing_chiptool.txt)

sayondeep, 2023-08-07:
@albrownwood Can you please tell the commit id and branch you are using for esp-idf and esp-matter?

dhrishi, 2023-08-10:
@albrownwood Can you please provide the requested details?

albrownwood, 2023-08-10:
this was about 25 commits ago from now, I am now observing a different behavior with current commit. I will close this issue and re-open a new one after I can confirm I still have an issue with recent commit.

closed 2023-08-10 by albrownwood