espressif/esp-matter #561: commissioning issue of matter over wifi

gautam1joshi, 2023-08-07:
hello,
i am using esp32-c3 runnig example code of light and when i try to do commissioning using chip-tool linux ,it sometimes works or sometimes it failed and when i commission device successfully then i can not on off the led on device logs of on-off command are as follow:
chip-tool logs:
sudo ./chip-tool-debug onoff off 1 1
[1691401142.908425][15408:15408] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1691401142.908561][15408:15408] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1691401142.908582][15408:15408] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1691401142.908598][15408:15408] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1691401142.908662][15408:15408] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-rcPkjs)
[1691401142.908804][15408:15408] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691401142.908812][15408:15408] CHIP:DL: NVS set: chip-counters/reboot-count = 8 (0x8)
[1691401142.909043][15408:15408] CHIP:DL: Got Ethernet interface: enp1s0
[1691401142.909183][15408:15408] CHIP:DL: Found the primary Ethernet interface:enp1s0
[1691401142.909344][15408:15408] CHIP:DL: Got WiFi interface: wlp3s0
[1691401142.910768][15408:15408] CHIP:DL: Found the primary WiFi interface:wlp3s0
[1691401142.910795][15408:15408] CHIP:IN: UDP::Init bind&listen port=0
[1691401142.910808][15408:15408] CHIP:IN: UDP::Init bound to port=43339
[1691401142.910811][15408:15408] CHIP:IN: UDP::Init bind&listen port=0
[1691401142.910823][15408:15408] CHIP:IN: UDP::Init bound to port=43130
[1691401142.910825][15408:15408] CHIP:IN: BLEBase::Init - setting/overriding transport
[1691401142.910828][15408:15408] CHIP:IN: TransportMgr initialized
[1691401142.910836][15408:15408] CHIP:FP: Initializing FabricTable from persistent storage
[1691401142.910868][15408:15408] CHIP:TS: Last Known Good Time: 2023-06-01T07:38:45
[1691401142.910990][15408:15408] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xAFC85A69BE5EE75C, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1691401142.911374][15408:15408] CHIP:ZCL: Using ZAP configuration...
[1691401142.912654][15408:15408] CHIP:DL: Avahi client registered
[1691401142.912873][15408:15408] CHIP:CTL: System State Initialized...
[1691401142.912894][15408:15408] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1691401142.912902][15408:15408] CHIP:CTL: Setting attestation nonce to random value
[1691401142.912905][15408:15408] CHIP:CTL: Setting CSR nonce to random value
[1691401142.912948][15408:15410] CHIP:DL: CHIP task running
[1691401142.912973][15408:15410] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1691401142.913016][15408:15410] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1691401142.913021][15408:15410] CHIP:CTL: Setting attestation nonce to random value
[1691401142.913024][15408:15410] CHIP:CTL: Setting CSR nonce to random value
[1691401142.915580][15408:15410] CHIP:CTL: Generating NOC
[1691401142.918382][15408:15410] CHIP:FP: Validating NOC chain
[1691401142.928792][15408:15410] CHIP:FP: NOC chain validation successful
[1691401142.928800][15408:15410] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1691401142.928803][15408:15410] CHIP:TS: Last Known Good Time: 2023-06-01T07:38:45
[1691401142.928805][15408:15410] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1691401142.928806][15408:15410] CHIP:TS: Retaining current Last Known Good Time
[1691401142.929442][15408:15410] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1691401142.929642][15408:15410] CHIP:TS: Committing Last Known Good Time to storage: 2023-06-01T07:38:45
[1691401142.929855][15408:15410] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: AFC85A69BE5EE75C)
[1691401142.931770][15408:15410] CHIP:TOO: Sending command to node 0x1
[1691401142.931859][15408:15410] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000001]
[1691401142.931863][15408:15410] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1691401142.931884][15408:15410] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 1 --> 2
[1691401142.931889][15408:15410] CHIP:DIS: Resolving AFC85A69BE5EE75C:0000000000000001 ...
[1691401142.932352][15408:15410] CHIP:DL: Avahi resolve found
[1691401142.932378][15408:15410] CHIP:DIS: Node ID resolved for AFC85A69BE5EE75C:0000000000000001
[1691401142.932380][15408:15410] CHIP:DIS: 	Hostname: 84F70366BE3C
[1691401142.932383][15408:15410] CHIP:DIS: 	IP Address #1: 192.168.6.103
[1691401142.932385][15408:15410] CHIP:DIS: 	Port: 5540
[1691401142.932387][15408:15410] CHIP:DIS: 	Mrp Interval idle: 600 ms
[1691401142.932389][15408:15410] CHIP:DIS: 	Mrp Interval active: 400 ms
[1691401142.932391][15408:15410] CHIP:DIS: 	TCP Supported: 1
[1691401142.932394][15408:15410] CHIP:DIS: Lookup clearing interface for non LL address
[1691401142.932408][15408:15410] CHIP:DIS: UDP:192.168.6.103%wlp3s0:5540: new best score: 2
[1691401142.932410][15408:15410] CHIP:DIS: Checking node lookup status after 0 ms
[1691401142.932412][15408:15410] CHIP:DIS: Keeping DNSSD lookup active
[1691401143.132035][15408:15410] CHIP:DIS: Checking node lookup status after 200 ms
[1691401143.132088][15408:15410] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:192.168.6.103:5540 while in state 2
[1691401143.132099][15408:15410] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
[1691401143.132150][15408:15410] CHIP:IN: SecureSession[0x7f6928009920]: Allocated Type:2 LSID:7664
[1691401143.132174][15408:15410] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000001
[1691401143.145191][15408:15410] CHIP:EM: <<< [E:50595i S:0 M:218439711] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1691401143.145234][15408:15410] CHIP:IN: (U) Sending msg 218439711 to IP address 'UDP:192.168.6.103:5540'
[1691401143.145389][15408:15410] CHIP:SC: Sent Sigma1 msg
[... 21 lines cut ...]
[1691401153.597471][15408:15408] CHIP:CTL: Shutting down the commissioner
[1691401153.597499][15408:15408] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1691401153.597515][15408:15408] CHIP:CTL: Shutting down the controller
[1691401153.597531][15408:15408] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1691401153.597546][15408:15408] CHIP:FP: Forgetting fabric 0x1
[1691401153.597570][15408:15408] CHIP:TS: Pending Last Known Good Time: 2023-06-01T07:38:45
[1691401153.597732][15408:15408] CHIP:TS: Previous Last Known Good Time: 2023-06-01T07:38:45
[1691401153.597747][15408:15408] CHIP:TS: Reverted Last Known Good Time to previous value
[1691401153.597770][15408:15408] CHIP:CTL: Shutting down the commissioner
[1691401153.597780][15408:15408] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1691401153.597793][15408:15408] CHIP:CTL: Shutting down the controller
[1691401153.597805][15408:15408] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1691401153.598787][15408:15408] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1691401153.598843][15408:15408] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1691401153.598869][15408:15408] CHIP:FP: Shutting down FabricTable
[1691401153.598886][15408:15408] CHIP:TS: Pending Last Known Good Time: 2023-06-01T07:38:45
[1691401153.598974][15408:15408] CHIP:TS: Previous Last Known Good Time: 2023-06-01T07:38:45
[1691401153.598983][15408:15408] CHIP:TS: Reverted Last Known Good Time to previous value
[1691401153.599171][15408:15408] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-YP5nap)
[1691401153.599603][15408:15408] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691401153.599653][15408:15408] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1691401153.599672][15408:15408] CHIP:DL: Inet Layer shutdown
[1691401153.599685][15408:15408] CHIP:DL: BLE shutdown
[1691401153.599699][15408:15408] CHIP:DL: System Layer shutdown
[1691401153.599891][15408:15408] CHIP:TOO: Run command failure: src/protocols/secure_channel/CASESession.cpp:485: CHIP Error 0x00000032: Timeout

dhrishi, 2023-08-07:
@gautam1joshi 

> it sometimes works or sometimes it failed 

This generally should not happen

> and when i commission device successfully then i can not on off the led on device logs of on-off

Is the commissioning complete and successful? Please upload (and not paste) the logs from the device side as well as chip-tool side for commissioning as well as control to debug this further.

gautam1joshi, 2023-08-07:
[chiptool_log.txt](https://github.com/espressif/esp-matter/files/12274959/chiptool_log.txt)
[esp.txt](https://github.com/espressif/esp-matter/files/12274961/esp.txt)

yes here it is.

dhrishi, 2023-08-07:
@gautam1joshi The device logs show that the node id being advertised for operational discovery is 1:

```
I (21899) chip[DIS]: Advertise operational node AFC85A69BE5EE75C-0000000000000001
I (21909) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: AFC85A69BE5EE75C-0000000000000001.

```

But, from the chip-tool, you are using node id 5

```
sudo ./chip-tool-debug binding read binding 5 1
```

Am I missing anything?

gautam1joshi, 2023-08-07:
sorry for mistake, i uploaded wrong file
this is the right one:
[chip-tool.txt](https://github.com/espressif/esp-matter/files/12278153/chip-tool.txt)

dhrishi, 2023-08-07:
So the commissioning is successful. Now you should control using node-id 1 from chip-tool. I would suggest you to redo the following steps:

1. Reset to factory ESP32-C3 : matter device factoryreset on the console
2. Commission using chip-tool (pairing ble-wifi command) with node id 1
3. Control (on/off/toggle) using chip-tool using node id 1

gautam1joshi, 2023-08-07:
it works after i reset the esp but when i try to do binding step i got error of unsupported cluster so is that mean binding is not supported or something i missing.
this error i get:
[binding_chip_tool.txt](https://github.com/espressif/esp-matter/files/12279004/binding_chip_tool.txt)

dhrishi, 2023-08-07:
>it works after i reset the esp but after sometimes it once again not work

This should not happen.

>and when i try to do binding step i got error of unsupported cluster so is that mean binding is not supported or something i missing.

Binding cluster server is present when there is a client cluster on the endpoint. If you are using the light example, then it won't be present

gautam1joshi, 2023-08-07:
> This should not happen.

okay so what could be a problem in commissioning?

>  Binding cluster server is present when there is a client cluster on the endpoint. If you are using the light example, then it won't be present

in that i am using another board which has matter switch code so i am trying to bind them so is that possible?

dhrishi, 2023-08-07:
>okay so what could be a problem in commissioning?

There should be no problem ideally. Once commissioned, it should work fine. If it fails, please upload the complete logs -> commissioning and then all the control commands for both chip-tool and the device.

>in that i am using another board which has matter switch code so i am trying to bind them so is that possible?

Please follow this: https://github.com/espressif/esp-matter/tree/main/examples/light_switch#2-post-commissioning-setup

gautam1joshi, 2023-08-08:
[esp.txt](https://github.com/espressif/esp-matter/files/12289517/esp.txt)
[chip-tool.txt](https://github.com/espressif/esp-matter/files/12289518/chip-tool.txt)
[chip_tool_command.txt](https://github.com/espressif/esp-matter/files/12289527/chip_tool_command.txt)
this are the logs before reset the board

gautam1joshi, 2023-08-08:
hello,
i once again try to do binding and this time it bind successfully but when switch try to send command it show error like this:

**E: 413624 [SVR]Failed to establish connection to node 0x0000000000000006**

shubhamdp, 2023-08-10:
@gautam1joshi I just want to confirm few things that you are trying...
1. Commission the light
2. commission the switch
3. Set the access control on the light
4. Bind the light to Switch
5. Press the button on switch and Light should toggle.

Logs from a single session would really help.. I would suggest please attach logs from light, switch, and chip-tool(along with commands used). So, that we can better help you here, or try to reproduce the issue.

dhrishi, 2023-08-27:
Closing this now. Please re-open by providing the necessary logs

closed 2023-08-27 by dhrishi