espressif/esp-matter #562: How can I bind the Matter Light to the Light Switch on Zigbee to Matter Bridge?

LongND0105, 2023-08-08:
I have a Matter Light application is running on ESP32. And I have a Light Switch on Zigbee to Matter Bridge. Can I bind Light to Light Switch? I have tried bind but chip-tool return CONSTRAINT_ERROR.
I defined Light Switch end point belike:
[image]

[image]

dhrishi, 2023-08-08:
@LongND0105 Please upload the chip-tool log for the commands that you use for the commissioning, ACL, and binding

LongND0105, 2023-08-08:
> @LongND0105 Please upload the chip-tool log for the commands that you use for the commissioning, ACL, and binding

These are the commands I used: node id of Light: 112, node id of bridge: 111

- Step 1: `accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": [111], "targets": null }]' 112 0`
- Step 2: `binding write binding '[{"fabricIndex": 1, "node":112, "endpoint":1, "cluster":6}]' 111 3`

dhrishi, 2023-08-10:
@LongND0105 Can you please check if your usecase is similar to https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light#esp-now-matter-bridge-light. In this example, we bind the esp-now button (presented as light switch in Matter) through the bridge to a Matter light.

dhrishi, 2023-08-14:
Hi @LongND0105 Did this help?

dhrishi, 2023-08-23:
@LongND0105 Please close if this helped and solved your issue

dhrishi, 2023-08-27:
Closing this now. Please re-open in case of anything

closed 2023-08-27 by dhrishi