espressif/esp-matter #563: Customize instance, hostname, mDNS?

law-ko, 2023-08-09:
Hello,

Currently, the instance/hostname/mDNS are defined in Matter service. Where can we modify those?

Thank you

wqx6, 2023-08-09:
The hostname, instance name, service name, and protocol name are restricted by SPEC(4.3 Discovery) for matter device. So you can not change the instance name or host name for matter mDNS service but you can add additional PTR/SRV/TXT record under  the matter mDNS hostname, you can use the APIs from `connectedhomeip/src/lib/dnssd/minimal_mdns` if you want to use the minimal mDNS implemented in `connectedhomip` repo. But we would suggest you to use platform mDNS by disabling minimal mDNS in menuconfig.

law-ko, 2023-08-10:
@wqx6  I have disabled `minimal_mdns` and attempt to set TXT record with `mdns_service_txt_set` after `esp_matter::start()` but it results in crash. 

```
    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Add mDNS TXT record under HTTP TCP */
    ESP_ERROR_CHECK(mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0));
    mdns_txt_item_t serviceTxtData[3] = {
        {"board","{esp32}"},
        {"u","user"},
        {"p","password"}
    };
    //set txt data for service (will free and replace current data)
    ESP_ERROR_CHECK(mdns_service_txt_set("_http", "_tcp", serviceTxtData, 3));
```

Crash log:
```
ESP0x400d926b: app_main at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/../main/app_main.cpp:440 (discriminator 1)

file: "../main/app_main.cpp" line 433
func: void app_main()
expression: mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0)

abort() was called at PC 0x40093c0b on core 0
0x40093c0b: _esp_error_check_failed at /home/espuser/esp/esp-idf/components/esp_system/esp_err.c:47

Backtrace: 0x4008226e:0x3ffbba10 0x40093c15:0x3ffbba30 0x4009a0a2:0x3ffbba50 0x40093c0b:0x3ffbbac0 0x400d926b:0x3ffbbaf0 0x401e2511:0x3ffbbc40 0x40096c91:0x3ffbbc80
0x4008226e: panic_abort at /home/espuser/esp/esp-idf/components/esp_system/panic.c:423

0x40093c15: esp_system_abort at /home/espuser/esp/esp-idf/components/esp_system/esp_system.c:153

0x4009a0a2: abort at /home/espuser/esp/esp-idf/components/newlib/abort.c:38

0x40093c0b: _esp_error_check_failed at /home/espuser/esp/esp-idf/components/esp_system/esp_err.c:47

0x400d926b: app_main at /home/espuser/esp-git/esp-matter/examples/im/outlet/build/../main/app_main.cpp:440 (discriminator 1)

0x401e2511: main_task at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x40096c91: vPortTaskWrapper at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

```

wqx6, 2023-08-10:
Please ensure that you call the platform_mdns APIs after the platform_mdns is initialized. The initialization will post the event `chip::DeviceLayer::DeviceEventType::kDnssdInitialized`.

law-ko, 2023-08-10:
@wqx6 Is there some type of `xEventGroupWaitBit` that we can wait on `chip::DeviceLayer::DeviceEventType::kDnssdInitialized` is true?

Currently thinking something like this..
```
while (1)
    {
        if (chip::DeviceLayer::DeviceEventType::kDnssdInitialized)
        {
            ESP_ERROR_CHECK(mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0));
            mdns_txt_item_t serviceTxtData[3] = {
                {"board","{esp32}"},
                {"u","user"},
                {"p","password"}
            };
            //set txt data for service (will free and replace current data)
            ESP_ERROR_CHECK(mdns_service_txt_set("_http", "_tcp", serviceTxtData, 3));
            break;
        }
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
```

wqx6, 2023-08-11:
@law-ko You can add use the registered event handler `app_event_cb` in `app_main.cpp` to handle the `chip::DeviceLayer::DeviceEventType::kDnssdInitialized` event.
```
static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
    case chip::DeviceLayer::DeviceEventType::kDnssdInitialized:
        ESP_ERROR_CHECK(mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0));
        ...
        ESP_ERROR_CHECK(mdns_service_txt_set("_http", "_tcp", serviceTxtData, 3));
        break;
    }
}
```

wqx6, 2023-08-14:
Could you please move the add service code under `chip::DeviceLayer::DeviceEventType::kCommissioningComplete` event. Or you need to set the mdns server hostname before you add service.

law-ko, 2023-08-14:
@wqx6 Do you mean to have
```
mdns_txt_item_t serviceTxtData[] = {
            {"board", "esp32"},
            {"path", "/"}
        };

        ESP_ERROR_CHECK(mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData,
                                        sizeof(serviceTxtData) / sizeof(serviceTxtData[0])));
        ESP_LOGW(TAG, "mDNS TXT Record set");
        break;
```
in both `chip::DeviceLayer::DeviceEventType::kCommissioningComplete` and `chip::DeviceLayer::DeviceEventType::kDnssdInitialized` case? But the `chip::DeviceLayer::DeviceEventType::kDnssdInitialized` will still crash.

Also I think you mentioned due to Matter specifications, we cannot define our own desired hostname. Then what do you mean by this:

> Or you need to set the mdns server hostname before you add service.

I think calling ` mdns_hostname_set("my-esp32");` would overwrite Matter defined hostname.

wqx6, 2023-08-15:
>in both chip::DeviceLayer::DeviceEventType::kCommissioningComplete and chip::DeviceLayer::DeviceEventType::kDnssdInitialized case? 

Only in `chip::DeviceLayer::DeviceEventType::kCommissioningComplete`

> I think calling  mdns_hostname_set("my-esp32"); would overwrite Matter defined hostname.

You can construct the host name according to **SPEC 4.3.1.1. Host Name Construction** and then set it.

law-ko, 2023-08-15:
@wqx6 `chip::DeviceLayer::DeviceEventType::kCommissioningComplete` case will not be triggered if commissioning is complete and then ESP32 reboots.

I think I am still having some issues with `mdns_service_add` with `ESP_ERR_INVALID_ARG` return:
```case chip::DeviceLayer::DeviceEventType::kDnssdInitialized:
        {
            ESP_ERROR_CHECK(mdns_hostname_set("my-esp32")); // added to ensure this is not the issue
            mdns_txt_item_t serviceTxtData[] = {
                {"board", "esp32"},
                {"path", "/"}
            };
            ESP_ERROR_CHECK(mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData,
                                            sizeof(serviceTxtData) / sizeof(serviceTxtData[0])));
            ESP_LOGW(TAG, "mDNS TXT Record set");
            break;
        }
```
Log:
```
E (18290) chip[SVR]: Server initialization complete
I (18290) chip[DIS]: Updating services using commissioning mode 0
I (18300) chip[DIS]: CHIP minimal mDNS started advertising.
I (18310) chip[DIS]: Advertise operational node 84CE2EA06AA86EEF-BB9F56397A918490
I (18320) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 84CE2EA06AA86EEF-BB9F56397A918490.
I (18330) chip[DIS]: mDNS service published: _matter._tcp
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400d8ecb
0x400d8ecb: app_event_cb(chip::DeviceLayer::ChipDeviceEvent const*, int) at /Users/user/esp-git/esp-matter/examples/test/outlet/main/app_main.cpp:183 (discriminator 1)

file: "./main/app_main.cpp" line 178
func: void app_event_cb(const chip::DeviceLayer::ChipDeviceEvent*, intptr_t)
expression: mdns_hostname_set("my-esp32")

abort() was called at PC 0x40093c0b on core 1
```

The same error happens also with putting `mdns_service_add` `chip::DeviceLayer::DeviceEventType::kCommissioningComplete` case

wqx6, 2023-08-15:
The `mdns_hostname_set()` should be call after `mdns_init()` so it should be call after `chip::DeviceLayer::DeviceEventType::kDnssdInitialized` event is posted and `mdns_service_add()` should be call after `mdns_hostname_set()`. So you can do the following things after receive the `chip::DeviceLayer::DeviceEventType::kDnssdInitialized` event
- set the mdns_hostname according **SPEC 4.3.1.1. Host Name Construction**
- add the mdns service

```
static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
    case chip::DeviceLayer::DeviceEventType::kDnssdInitialized:
        char host_name[32];
        uint8_t eth_mac[6];
        esp_wifi_get_mac(WIFI_IF_STA, eth_mac);
        snprintf(host_name, 32, "%02x%02x%02x%02x%02x%02x.local", eth_mac[0], eth_mac[1], eth_mac[2], eth_mac[3],
                     eth_mac[4], eth_mac[5]);
        ESP_ERROR_CHECK(mdns_hostname_set(host_name));
        ESP_ERROR_CHECK(mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0));
        ...
        ESP_ERROR_CHECK(mdns_service_txt_set("_http", "_tcp", serviceTxtData, 3));
        break;
    }
}
```

law-ko, 2023-08-16:
@wqx6 Tried with disabling minimal mDNS but results in earlier crash before commissioning, crash at the init stage:
```
Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
W (3461) app_main: Dnssd initialized
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400014fd  PS      : 0x00060630  A0      : 0x80001459  A1      : 0x3ffef960  
A2      : 0xe425754c  A3      : 0xe4257548  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffef930  
A10     : 0x3ffdfa94  A11     : 0x3fff4b70  A12     : 0x00060021  A13     : 0xfffffffd  
A14     : 0x3ffe43b8  A15     : 0x0000abab  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xe425754c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fa:0x3ffef960 0x40001456:0x3ffef970 0x40001445:0x3ffef990 0x401295d1:0x3ffef9b0 0x4012daf7:0x3ffef9f0 0x4012db9a:0x3ffefa40 0x400d8e65:0x3ffefa60 0x401d818d:0x3ffefad0 0x4013e63d:0x3ffefaf0 0x4013e91a:0x3ffefb10 0x4013e94d:0x3ffefb80 0x40096c91:0x3ffefba0
0x401295d1: _mdns_allocate_txt at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/managed_components/espressif__mdns/mdns.c:2491

0x4012daf7: _mdns_allocate_txt at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/managed_components/espressif__mdns/mdns.c:2484
 (inlined by) mdns_service_txt_set_for_host at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/managed_components/espressif__mdns/mdns.c:5996

0x4012db9a: mdns_service_txt_set at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/managed_components/espressif__mdns/mdns.c:6025

0x400d8e65: app_event_cb(chip::DeviceLayer::ChipDeviceEvent const*, int) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/main/app_main.cpp:179 (discriminator 2)

0x401d818d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:336

0x4013e63d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:301

0x4013e91a: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:505
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257

0x4013e94d: chip::DeviceLayer::PlatformManager::RunEventLoop() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:405
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280

0x40096c91: vPortTaskWrapper at /Users/lawrenceko/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

I have added `idf.py add-dependency "espressif/mdns^1.2.0"` mDNS dependency and include with `#include "mdns.h"`

wqx6, 2023-08-17:
Please ensure the txt entries are valid when you call `mdns_service_txt_set` as the log shows the `_mdns_allocate_txt` fails

closed 2023-08-18 by law-ko

PhLuReh, 2024-02-12:
Is there a way to change the netif hostname, which get's tracked by the DHCP? I found the [esp-idf documentation](https://esp32tutorials.com/set-esp32-hostname-esp-idf/), but obviously I'm not able to simple create or get the proper netif. Can you give advice, if this is possible in esp-matter?