espressif/esp-matter #564: Stable version of ESP-IDF

aashikareliya, 2023-08-09:
I am Using ESP32-C3 and i want to flash Light example from esp-matter SDK using ESP-IDF. for that which version of ESP-IDF is Stable ?

shubhamdp, 2023-08-09:
Documentation suggests esp-idf v5.0.1 for ESP32-C3, https://github.com/espressif/esp-matter/#supported-esp-idf-and-connectedhomeip-versions.

I'd suggest you use [v5.1](https://github.com/espressif/esp-idf/releases/tag/v5.1), that is the latest stable esp-idf release.

aashikareliya, 2023-08-09:
thank you for info. @shubhamdp 
which is the android application for ble commissioning of matter device (esp32-c3) other than chip tool and GHSMA?

dhrishi, 2023-08-09:
You can use the Google Home app (with a hub in your home). Closing this as the original question is answered

closed 2023-08-09 by dhrishi