espressif/esp-matter #566: Error: Unable to detect the device in ESP Lanchpad

varshilgandhi, 2023-08-09:
Hi, Varshil here,
I am trying to flash the light bulb example code in ESP-C3 Board.
When I connect the Dev kit of ESP-C3 with the Laptop and then click on the connect button of ESP Launchpad.
When I click on the connect button it gives me unable to detect the device.
BUT I already connect board with system and for cross checking I type this command in terminal:
ls /dev/tty* 
and after run above code I can see my board is connected.
But ESP Launchpad shows that it is unable to detect.

dhrishi, 2023-08-09:
@varshilgandhi Please raise the issue in https://github.com/espressif/esp-launchpad for better assistance

closed 2023-08-09 by dhrishi

varshilgandhi, 2023-08-10:
@dhrishi Okay Thank You.