espressif/esp-matter #568: Matter esp32c3 commissioning

aashikareliya, 2023-08-10:
I am using ESP32C3 with matter Ligt Example code. and when I try to perform commissioning in Interactive mode using command:
**chip-tool interactive start**
It is showing error of 
**CHIP:CTL: Init Storage failure: ../examples/chip-tool/third_party/connectedhomeip/src/controller/ExamplePersistentStorage.cpp:70: Error 0x000000AD**

How can I resolve this?

dhrishi, 2023-08-10:
@aashikareliya Please provide sufficient information. Was the `install.sh` and `export.sh` in esp-matter successful? Also, details about your setup - OS, VMs or anything else. Does the chip-tool in non-interactive mode work?

dhrishi, 2023-08-11:
Please re-open after providing all the relevant information as mentioned above

closed 2023-08-11 by dhrishi