espressif/esp-matter #570: How to reset matter? factory_reset() not working as expected

emmby, 2023-08-12:
I'd like to reset matter when the user hits a button to enable them to reprovision onto a new matter network.

I tried calling esp_matter::factory_reset() per the [app_reset example](https://github.com/espressif/esp-matter/blob/a84ee2a2463e253fddf87a62a0b4236fb547d8a4/examples/common/app_reset/app_reset.cpp#L32), but on next boot `chip::DeviceLayer::ConfigurationMgr().IsFullyProvisioned()` is still returning true.

What is the proper way to reset matter to enable reprovisioning?

emmby, 2023-08-12:
Switching to chip::DeviceLayer::ConfigurationMgr().InitiateFactoryReset() seems to solve the problem

shubhamdp, 2023-08-14:
@emmby `esp_matter::factory_reset()` performs cleanup from esp-matter SDK and calls [`ConfigurationMgr().InitiateFactoryReset()`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1063). Any specific thing that is blocking the reset?

emmby, 2023-08-14:
Ah, the reason `esp_matter::factory_reset()` wasn't working for me is that it apparently was dumping core:

```
08:24:35.911 -> Factory resetting on button hold
08:24:36.218 -> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
08:24:36.218 -> 
08:24:36.218 -> Core  1 register dump:
08:24:36.218 -> PC      : 0x4009a266  PS      : 0x00060735  A0      : 0x80098c4e  A1      : 0x3ffbf1bc  
08:24:36.218 -> A2      : 0x3ffb6e98  A3      : 0x3ffbd0b4  A4      : 0x00000004  A5      : 0x00060723  
08:24:36.218 -> A6      : 0x00060723  A7      : 0x00000001  A8      : 0x3ffbd0b4  A9      : 0x00000001  
08:24:36.218 -> A10     : 0x3ffbd0b4  A11     : 0x00000001  A12     : 0x00000020  A13     : 0x3ffbf478  
08:24:36.251 -> A14     : 0x007bf5e8  A15     : 0x003fffff  SAR     : 0x0000000c  EXCCAUSE: 0x00000006  
08:24:36.251 -> EXCVADDR: 0x00000000  LBEG    : 0x40086085  LEND    : 0x4008608d  LCOUNT  : 0x00000027  
08:24:36.251 -> Core  1 was running in ISR context:
08:24:36.251 -> EPC1    : 0x400e4c53  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
08:24:36.251 -> 
08:24:36.283 -> 
08:24:36.283 -> Backtrace: 0x4009a263:0x3ffbf1bc |<-CORRUPTED
08:24:36.283 -> 
08:24:36.283 -> 
08:24:36.283 -> Core  0 register dump:
08:24:36.283 -> PC      : 0x4009a3fb  PS      : 0x00060f35  A0      : 0x80097a18  A1      : 0x3ffb7230  
08:24:36.283 -> A2      : 0x3ffb6ec0  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060f23  
08:24:36.283 -> A6      : 0x00060f20  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0xffffffff  
08:24:36.283 -> A10     : 0x00000003  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x00060f23  
08:24:36.316 -> A14     : 0x007b6ec0  A15     : 0x003fffff  SAR     : 0x00000001  EXCCAUSE: 0x00000006  
08:24:36.316 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
08:24:36.316 -> 
08:24:36.316 -> 
08:24:36.316 -> Backtrace: 0x4009a3f8:0x3ffb7230 0x40097a15:0x3ffb7270 0x4008e6fe:0x3ffb72b0
08:24:36.316 -> 
08:24:36.316 -> 
08:24:36.316 -> 
08:24:36.316 -> 
08:24:36.316 -> ELF file SHA256: 3a814a1e502810a0
08:24:36.316 -> 
08:24:36.525 -> Rebooting...
```

Here are the versions I'm using:

[esp32-arduino-matter-builder](https://github.com/Yacubane/esp32-arduino-matter-builder)	v1.0.0-beta6
[Matter](https://github.com/project-chip/connectedhomeip)	08b13661b6
[esp-matter](https://github.com/espressif/esp-matter)	603296a
[esp-idf](https://github.com/espressif/esp-idf)	v4.4.5
[arduino-esp32](https://github.com/espressif/arduino-esp32)	2.0.11

emmby, 2023-08-16:
Sure, I'm happy to generate a backtrace for you if you can tell me how (especially using arduino tools).

Here's a diff of my code change:
[image]

The code in pink does not crash, the code in green crashes with the logs in my previous comment.

shubhamdp, 2023-08-21:
I see all your changes in pink other than the commented line. Is `esp_matter::factory_reset()` crashing? and `chip::DeviceLayer::ConfigurationMgr().InitiateFactoryReset()` not?

If yes, then can you please add some debug prints in [`factorty_reset()`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1035) and see where it is crashing?

> 08:24:36.316 -> Backtrace: 0x4009a3f8:0x3ffb7230 0x40097a15:0x3ffb7270 0x4008e6fe:0x3ffb72b0

Backtrace contains PC:SP pairs. For mapping this backtrace to the code lines, you can use add2line with the generated `.elf` file. (you may have to look up to find the location of .elf file).
```
addr2line -e path/to/file.elf <space separated list of PCs>
```

shubhamdp, 2023-08-30:
@emmby were you able to collect the backtrace?

dhrishi, 2023-09-30:
@emmby If you are still seeing this issue, then please share the backtrace as requested

closed 2023-09-30 by dhrishi