espressif/esp-matter #571: Possible to create matter hub to control matter endpoint?

ahmed8812, 2023-08-14:
Hi,

Would it be possible to create own matter hub that can control matter endpoint?

For either Homekit, Google Home or Alexa I would require respective manufacturer's hub to control Matter devices.
But without having respective manufacturer's hub (Homepod etc) connected, does Matter supports creating our own Matter hub that only by utilising HomeApp, Alexa app, Google app on phone be able to control our own hub like sending off/on, control light colours etc to Matter certified endpoint?

On the side note, I have tested 'Matter BLE Mesh Bridge' on Google and Homekit to try to control non-Matter endpoint. Seems this product is not yet supported by both manufacturers.

shubhamdp, 2023-08-16:
@ahmed8812 Are you also asking, if you can replace the Google/Apple/Amazon's hub with your hub and continue to use the Home apps from them with your hub. If yes, I think this is not possible right now.

If you want to develop your hub (say running on ESP32) and your phone app then that is absolutely possible.

If you want to just develop a phone app which controls Matter devices that is also possible. (This is without hub)

ahmed8812, 2023-08-16:
@shubhamdp Thanks for replying

Say if I want to make hub with my own app, this only means that only that app can control the hub correct?

Is it possible like to have my own matter hub, pair to any platform (Apple, Google, Alexa) that runs together  with their respective hubs, that whoever uses any platform can still control my hub?
For now I found out by using matter BLE mesh bridge example while using HomePod and Google Nest, both platfrom cannot still support my own hub

ahmed8812, 2023-08-22:
Hi @shubhamdp ,

Ok let me explain the use case:

For example the Matter hub that will be created able to control other Matter's endpoint.
Which means, using our designed Matter app, I can pair the hub to the app, and bind the hub and matter Light together, the hub can control the Matter light without any intervention of Apple / Alexa/ Google hubs
Similarly, if user pair the hub to any platform (Google/Alexa/Apple), user can still bind other Matter endpoints to the hub and do controls. 
If this technically sound to achieve according to Matter?

For the Matter BLE mesh bridge example that I run, it can bind successfully with the non Matter mesh device.
Except when pair to platform Apple, Google this error appears in their respective app. Please see the images

[image]
[image]

Thanks

shubhamdp, 2023-08-24:
Ahh, now I get it, your hub is nothing but a Matter node with client cluster, i.e. controller. You can commission the hub to ecosystems and use binding cluster for binding the other Matter nodes with your hub.

If you take a look at [light_switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) example, it implements the client cluster which we can then bind with eg: light.

I found one more example in our esp-box repository which uses the ESP32-S3 Box as a controller, https://github.com/espressif/esp-box/tree/master/examples/matter_switch.

In a nutshell, what have you asked is achievable. The word `HUB` got me confused.

ahmed8812, 2023-08-24:
Hi

Thanks I will look into it

Can you also see why from the images I sent, the Google app and Home app do
not recognised the BLE mesh bridge matter sample that I run? Is it due that
the platforms has not yet supported to run matter bridge through their app?

Thanks

shubhamdp, 2023-08-30:
> Thanks I will look into it

Any update on this?

> Can you also see why from the images I sent, the Google app and Home app do
not recognised the BLE mesh bridge matter sample that I run? Is it due that
the platforms has not yet supported to run matter bridge through their app?

Home apps does support the bridge devices, I remember testing that and it does shows up in there.
Can you share more details about app version, hub version, esp-idf and esp-matter commit.

dhrishi, 2023-09-18:
Closing this now

closed 2023-09-18 by dhrishi