espressif/esp-matter #576: Dynamically add and remove endpoints after Matter start

law-ko, 2023-08-17:
Hello,

Is it possible to create and remove endpoints after `esp_matter::start(app_event_cb);`?

We have tried to bind `nightlight_endpoint = dimmable_light::create(matter_node, &nightlight_config, ENDPOINT_FLAG_NONE, nullptr);` by defining `matter_node` as a global variable, and the `nightlight_endpoint` does get a valid endpoint with `nightlight_endpoint_id = endpoint::get_id(nightlight_endpoint);`. But when scanning with Apple Home app, the newly added endpoints do not show after commissioning.

Thank you

jadhavrohit924, 2023-08-17:
@law-ko Basically `matter start` creates data model of the device type so, you have to create device tyep before `matter start`.

law-ko, 2023-08-17:
@jadhavrohit924 Is there any way to `matter stop` instead of restarting ESP32?

jadhavrohit924, 2023-08-17:
What do you mean by matter stop? Could you please explain a bit more?

law-ko, 2023-08-17:
@jadhavrohit924 We have `esp_matter::start` function to call, is there a function like `esp_matter::stop` for us to stop the Matter service and then add/remove the endpoints and call `esp_matter::start` again?

jadhavrohit924, 2023-08-17:
@law-ko No, there is no such API as of now.

law-ko, 2023-08-17:
@jadhavrohit924 Is there any function call to make endpoints or clusters show as offline or error detected?

jadhavrohit924, 2023-08-18:
There are `get` functions for endpoint and cluster which returns endpoint and cluster handel, respectively. Please check this file https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h

law-ko, 2023-08-18:
@jadhavrohit924 If we `destroy` the endpoint to indicate that the cluster or endpoint is offline, can we add it back with `create` or `resume` without restarting device (i.e. `esp_matter::start()`)?

jadhavrohit924, 2023-08-22:
@law-ko We can add new endpoint anytime after matter start. Spec is not restricting us to not add/remove endpoints, please check this https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/System-Model.adoc#21-dynamic-endpoint-allocation . 
To add new endpoint after matter start make sure that you enable the endpoint.
```
fan::config_t fan_config;
endpoint_t *endpoint1 = fan::create(node1, &fan_config, ENDPOINT_FLAG_NONE, NULL);

endpoint::enable(endpoint1);
```

To destroy endpoint make sure you have set `ENDPOINT_FLAG_DESTROYABLE` while creating endpoint.

law-ko, 2023-08-23:
@jadhavrohit924 Thank you. With using `destroy` and `enable` I can dynamically add/remove endpoints after `esp_matter::start()`. To clarify, what's the difference between using `disable` and `destroy`?

Disable: `endpoint::disable(plug_endpoint);`
Destroy: `endpoint::destroy(matter_node, plug_endpoint);`

jadhavrohit924, 2023-08-23:
Please check the API references [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h)

jadhavrohit924, 2023-08-25:
Please close the issue if your problem is solved, thanks.

closed 2023-08-25 by law-ko