espressif/esp-matter #577: ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x420c0235

likenamehaojie, 2023-08-17:
0x420c0235: chip::Shell::streamer_esp32_init(chip::Shell::streamer const*) at /Users/like/Documents/esp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/shell/streamer_esp32.cpp:78 (discriminator 1)

file: "src/lib/shell/streamer_esp32.cpp" line 101
func: int chip::Shell::streamer_esp32_init(streamer_t*)
expression: esp_console_init(&console_config)

abort() was called at PC 0x40380fb3 on core 0
0x40380fb3: _esp_error_check_failed at /Users/like/Documents/esp/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x40375be6:0x3fcb3fe0 0x40380fbd:0x3fcb4000 0x4038716a:0x3fcb4020 0x40380fb3:0x3fcb4090 0x420c0235:0x3fcb40c0 0x420f230f:0x3fcb4130 0x420b90a2:0x3fcb4150 0x420341f6:0x3fcb4170 0x42009640:0x3fcb41a0 0x420f5cdf:0x3fcb4260 0x4038359d:0x3fcb4290
0x40375be6: panic_abort at /Users/like/Documents/esp/esp-idf/components/esp_system/panic.c:452

0x40380fbd: esp_system_abort at /Users/like/Documents/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93

0x4038716a: abort at /Users/like/Documents/esp/esp-idf/components/newlib/abort.c:38

0x40380fb3: _esp_error_check_failed at /Users/like/Documents/esp/esp-idf/components/esp_system/esp_err.c:50

0x420c0235: chip::Shell::streamer_esp32_init(chip::Shell::streamer const*) at /Users/like/Documents/esp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/shell/streamer_esp32.cpp:78 (discriminator 1)

0x420f230f: chip::Shell::streamer_init(chip::Shell::streamer const*) at /Users/like/Documents/esp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/shell/streamer.cpp:40

0x420b90a2: chip::Shell::Engine::Init() at /Users/like/Documents/esp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/shell/Engine.cpp:50

0x420341f6: esp_matter::console::init() at /Users/like/Documents/esp/esp-matter/components/esp_matter_console/esp_matter_console.cpp:142

0x42009640: app_main at /Users/like/Documents/esp/esp-matter/examples/light/main/app_main.cpp:198

0x420f5cdf: main_task at /Users/like/Documents/esp/esp-idf/components/freertos/app_startup.c:217 (discriminator 13)

0x4038359d: vPortTaskWrapper at /Users/like/Documents/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 1658a976a

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b04

dhrishi, 2023-08-18:
@likenamehaojie Please  provide details about when you encounter this issue. Does it happen everytime? Is this the light example without any change or do you have any changes (code/menuconfig options) on top of it? Also, details about the commit/branch of ESP-IDF and esp-matter along with the SoC used will also help

dhrishi, 2023-08-27:
Please re-open by providing the necessary details

closed 2023-08-27 by dhrishi