espressif/esp-matter #578: assert failed: esp_efuse_rtc_calib_get_chan_compens

sallah21, 2023-08-17:
Hi!
I use ESP32-C6 for my university project, so i'm new to ESP and Matter.
I tried to run some examples for esp-matter (light example).
After successful build I flashed it to my devkit and got panic abort as follows. 
Is it my mistake or firmware bug? For all help thanks in advance

I (23) boot: ESP-IDF v5.2-dev-823-g903af13e84-dirty 2nd stage bootloader
I (24) boot: compile time Aug 17 2023 12:31:43
I (25) boot: chip revision: v0.0
I (29) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (67) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (74) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (81) boot:  3 phy_init         RF data          01 01 00017000 00001000
I (89) boot:  4 ota_0            OTA app          00 10 00020000 001c0000
I (96) boot:  5 ota_1            OTA app          00 11 001e0000 001c0000
I (104) boot:  6 ot_storage       Unknown data     01 81 003a0000 00002000
I (112) boot: End of partition table
I (116) esp_image: segment 0: paddr=00020020 vaddr=42120020 size=3d150h (250192) map
I (183) esp_image: segment 1: paddr=0005d178 vaddr=40800000 size=02ea0h ( 11936) load
I (187) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=119b1ch (1153820) map
I (461) esp_image: segment 3: paddr=00179b44 vaddr=40802ea0 size=11e14h ( 73236) load
I (481) esp_image: segment 4: paddr=0018b960 vaddr=40814cc0 size=03778h ( 14200) load
I (490) boot: Loaded app from partition at offset 0x20000
I (491) boot: Disabling RNG early entropy source...
I (502) cpu_start: Unicore app
I (502) cpu_start: Pro cpu up.
W (513) clk: esp_perip_clk_init() has not been implemented yet
I (519) cpu_start: Pro cpu start user code
I (519) cpu_start: cpu freq: 160000000 Hz
I (520) cpu_start: Application information:
I (522) cpu_start: Project name:     light
I (527) cpu_start: App version:      v1.0
I (532) cpu_start: Compile time:     Aug 17 2023 12:31:33
I (538) cpu_start: ELF file SHA256:  fe21f1edd...
I (543) cpu_start: ESP-IDF:          v5.2-dev-823-g903af13e84-dirty
I (550) cpu_start: Min chip rev:     v0.0
I (555) cpu_start: Max chip rev:     v0.99 
I (560) cpu_start: Chip rev:         v0.0
I (565) heap_init: Initializing. RAM available for dynamic allocation:
I (572) heap_init: At 4082A390 len 00052280 (328 KiB): D/IRAM
I (578) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (585) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (592) spi_flash: detected chip: generic
I (596) spi_flash: flash io: dio
W (600) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (614) eFuse: calibration efuse version does not match, set default version to 0
W (642) eFuse: calibration efuse version does not match, set default version to 0

assert failed: esp_efuse_rtc_calib_get_chan_compens esp_efuse_rtc_calib.c:59 (version == ESP_EFUSE_ADC_CALIB_VER)
Stack dump detected
Core  0 register dump:
MEPC    : 0x4080045c  RA      : 0x4080a41e  SP      : 0x4087dff0  GP      : 0x408154c0  
0x4080045c: panic_abort at /home/salsa/esp/esp-idf/components/esp_system/panic.c:452

0x4080a41e: __ubsan_include at /home/salsa/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x4087e181  A0      : 0x4087e048  A1      : 0x40817669  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40829000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000056  S3      : 0x4087e048  
S4      : 0x4087e048  S5      : 0x420781de  S6      : 0x40800258  S7      : 0x00119b1c  
0x420781de: esp_efuse_rtc_calib_get_chan_compens at /home/salsa/esp/esp-idf/components/efuse/esp32c6/esp_efuse_rtc_calib.c:60

0x40800258: rv_utils_dbgr_is_attached at /home/salsa/esp/esp-idf/components/riscv/include/riscv/rv_utils.h:201
 (inlined by) esp_cpu_dbgr_is_attached at /home/salsa/esp/esp-idf/components/esp_hw_support/include/esp_cpu.h:500
 (inlined by) call_start_cpu0 at /home/salsa/esp/esp-idf/components/esp_system/port/cpu_start.c:302

S8      : 0x0003d150  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x4087e048 "assert failed: esp_efuse_rtc_calib_get_chan_compens esp_efuse_rtc_calib.c:59 (version == ESP_EFUSE_ADC_CALIB_VER)") at /home/salsa/esp/esp-idf/components/esp_system/panic.c:452
452	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x4087e048 "assert failed: esp_efuse_rtc_calib_get_chan_compens esp_efuse_rtc_calib.c:59 (version == ESP_EFUSE_ADC_CALIB_VER)") at /home/salsa/esp/esp-idf/components/esp_system/panic.c:452
#1  0x4080a41e in esp_system_abort (details=details@entry=0x4087e048 "assert failed: esp_efuse_rtc_calib_get_chan_compens esp_efuse_rtc_calib.c:59 (version == ESP_EFUSE_ADC_CALIB_VER)") at /home/salsa/esp/esp-idf/components/esp_system/port/esp_system_chip.c:90
#2  0x4080ebd8 in __assert_func (file=<optimized out>, file@entry=0x42139b3b "", line=<optimized out>, func=<optimized out>, expr=<optimized out>) at /home/salsa/esp/esp-idf/components/newlib/assert.c:81
#3  0x420781de in esp_efuse_rtc_calib_get_chan_compens (version=<optimized out>, adc_unit=<optimized out>, adc_channel=<optimized out>, atten=<optimized out>) at /home/salsa/esp/esp-idf/components/efuse/esp32c6/esp_efuse_rtc_calib.c:60
#4  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: fe21f1edd

sayondeep, 2023-08-18:
@sallah21 could you please specify the branch and commit id for esp-idf and esp-matter that you are using. It will help in reproducing the issue.

sallah21, 2023-08-18:
Of course 
IDF id: commit 903af13e847cd301e476d8b16b4ee1c21b30b5c6 (HEAD -> master, origin/master, origin/HEAD)
Merge: 0b5ab48437 873901e7aa
Author: Aditya Patwardhan <aditya.patwardhan@espressif.com>
Date:   Fri May 26 18:13:05 2023 +0800

ESP-matter id:commit 9be53266b1e934797b9a1cc08961551bc4661677 (grafted, HEAD -> main, origin/main, origin/HEAD)
Author: Hrishikesh Dhayagude <hrishi@espressif.com>
Date:   Tue May 30 23:09:21 2023 +0800

sayondeep, 2023-08-18:
@sallah21 please use the ESP-IDF commit id and branch as mentioned [here](https://github.com/espressif/esp-matter/tree/main#supported-esp-idf-and-connectedhomeip-versions) for esp32-C6.

sallah21, 2023-08-18:
Ok it worked. Thank's for help. :)

closed 2023-08-18 by sallah21

sayondeep, 2023-08-18:
It should work on the ESP-Matter main branch as well.