espressif/esp-matter #580: The requirements for OTA in Matter device?

law-ko, 2023-08-18:


dhrishi, 2023-08-18:
@law-ko Matter specification mentions that OTA is mandatory. It can either be Matter based OTA or manufacturer specific OTA.

closed 2023-08-18 by dhrishi

law-ko, 2023-08-18:
@dhrishi Matter based OTA means Requestor OTA that requires a Provider OTA device on the same network?

which means Matter based OTA does not have something that is similar to esp_http_ota?

dhrishi, 2023-08-18:
@law-ko Yes. Matter based OTA is the one defined in the specification, that requires a OTA requestor and OTA provider

>which means Matter based OTA does not have something that is similar to esp_http_ota?

That's right

law-ko, 2023-08-24:
@dhrishi Is there any way to show new OTA available in Apple Home app with HTTP-based OTA by Matter?