espressif/esp-matter #581: Matter bridge to expose ESP-Mesh devices?

erumnasir, 2023-08-18:
Is there anything that prevents one from making a Matter bridge with this SDK and expose compatible device types such as switches and fans to Matter? With a single ESP-32 module?

omarkamal, 2023-08-19:
I'm interested in knowing this too

shubhamdp, 2023-08-21:
@erumnasir If you talking about the [Matter Bridge for Non-Matter Devices](https://blog.espressif.com/matter-bridge-for-non-matter-devices-d3b7f003a004), then we have three bridge [examples](https://github.com/espressif/esp-matter/tree/main/examples) implemented (blemesh, esp-now, and zigbee) in the SDK.

If not, can you please share more details.

shubhamdp, 2023-08-28:
@erumnasir any update on this, if the above comment answers your question, please close the issue.

closed 2023-09-05 by dhrishi