espressif/esp-matter #583: problem with install docs on WSL - BEFORE you execute for matter download, for WSL, you HAVE to enforce case sensitivity

riwalker, 2023-08-19:
note:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup
“To clone the esp-matter repository with all the submodules, use the following command:”
git clone --recursive https://github.com/espressif/esp-matter.git

**BEFORE you execute for matter download, for WSL, you HAVE to enforce case sensitivity:**
https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity
failed:

[P107553 22:46:43.154 deployer.go:409 I] [unzip  7/12] Deployed fuchsia/third_party/clang/linux-amd64:AD59JKqO1TWgxp7uA0Fq99aIrVCXG2i4HShWi6sFFV8C in 135.6s
Errors:
  failed to install fuchsia/third_party/armgcc/linux-amd64 in "arm": adding files to the site root: lstat /mnt/c/Users/xyz/source/repos/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/0/9z1FnQUHn-LGwcPNIsxnZzTE3HjoopYzwWlK3f1-HG0C/share/doc/gdb/qMemTags.html: no such file or directory

sayondeep, 2023-08-21:
@riwalker It is okay to clone recursively using `git clone --recursive https://github.com/espressif/esp-matter.git` but for esp-matter only esp32 specific submodules are necessary which can be done by running './scripts/checkout_submodules.py --platform esp32 linux --shallow` from path/to/esp-matter/connectedhomeip/connectedhomeip as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories).
If you follow the above steps case-sensitivity is not-required for WSL.

riwalker, 2023-08-21:
I'm commenting on following the instructions exactly as provided on the official espressif documentation, that whilst running WSL2, the esp-matter install of esp-matter triggers an error, which is HIGHLY frustrating, and took some debug to figure out the issue.
we need Espressif to update your documation so others do not find the same problem.
once i change windows to accept the case-sensitivity, as per the microsoft url, and reboot, it worked fine

WRT the clone of "esp-matter repository with all the submodules" or "shallow clone" - who knows ? there is no text to say what is needed and what works - that's the frustration here.

sayondeep, 2023-08-21:
@riwalker It would really help to know the cause of the issue to update the documentation.
May I know if you cloned the esp-matter repo inside WSL or cloned it on Windows and accessing on WSL using `\mnt\ ...` i.e., by mounting.
The reason being that WSL is by default case-sensitive as mentioned [here](https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity).

riwalker, 2023-08-21:
I followed the [instructions ](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup)precisely
in section "2.1.1.1 Windows 10"
so installed and enabled WSL 2 as per the given [url](https://learn.microsoft.com/en-us/windows/wsl/install)
used all the standard distro (which is latest Ubuntu 22.04) and standard options.
ran uname -a to check linux version, then install usbipd
had prior installed vscode, to followed the updated [url](https://code.visualstudio.com/docs/remote/wsl)
and when installing the repos, i was from /mnmt/c/Users/xyz/source/repos

i had already completed the esp-idf ok, only the esp-matter threw the error

sayondeep, 2023-08-21:
Okay @riwalker . The issue is due to mounting of the directory and not cloning from inside the WSL. Thanks for the feedback. 

Regarding the cloning methodology, the commands mentioned [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories) clones only what is necessary for ESP-Matter to build . It is explicitly mentioned that it would shallow clone, but as I mentioned cloning recursively works as well.

closed 2023-10-25 by espressif-bot