espressif/esp-matter #584: Making http connections from a matter client in Arduino

emmby, 2023-08-21:
I have a matter client (written in Arduino using https://github.com/Yacubane/esp32-arduino-matter ) that works just fine and is able to reliably make network connections to sntp servers like pool.ntp.org.

However, when I try to make http connections to sites like www.google.com, I get a -1 connection refused. I suspect my problem is that the Arduino http client does not support ipv6: https://arduino.stackexchange.com/questions/93628/internet-in-esp32-is-not-working-with-ipv6

Has anyone successfully made an outbound http/s connection to google.com or similar using esp-matter on Arduino? Are there better ways to do it?

shubhamdp, 2023-08-28:
Are you saying you want to make an API call to google services using http library? Can you share the code snippet for what you are trying to do? I don't thing it has anything to do with IPv6. Also, what is the error code returned here, logs would be helpful, please attach them?

emmby, 2023-09-05:
Sorry, I don't have a code snippet handy. For posterity I rewrote arduino-esp32 [HttpUpdate](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdate/src/HTTPUpdate.h) to use esp_http_client
but I probably should have just used https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_https_ota.html directly.

closed 2023-09-05 by emmby