espressif/esp-matter #585: Proprietary update system, how to notify the Matter stack

albrownwood, 2023-08-22:
Hello,
when implementing a custom fw update system that is not using the OTA provider defined in Matter, are there APis the application running on the accessory can call to inform the matter stack that a fw update is available/in progress/completed?

albrownwood, 2023-08-30:
@shubhamdp sorry let me put it differently, the question is about being able to report the status of the proprietary fw update system (e.g. update available/ update in progress/ update completed) so it is displayed to the user in the iOS/Android native home app.

shubhamdp, 2023-08-30:
Ahh, I just replied in https://github.com/espressif/esp-matter/issues/600#issuecomment-1698566765

dhrishi, 2023-09-04:
@albrownwood Please close your issue if the question is answered

dhrishi, 2023-09-30:
Closing this now

closed 2023-09-30 by dhrishi