espressif/esp-matter #586: Thermostat definition

beckmx, 2023-08-22:
Hello guys, I have been trying to run this code to define a thermostat but looks like it doesnt work yet, at least the basics to me seem fine, when I sync with an iPhone i dont see any controls in the thermostat UI.

Here is my code:

```
#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <esp_matter.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>

#include <app_priv.h>
#include <app_reset.h>
#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
#include <platform/ESP32/OpenthreadLauncher.h>
#endif

#include <app/server/CommissioningWindowManager.h>
#include <app/server/Server.h>
#define CHIP_DEVICE_CONFIG_PRODUCT_NAME "my test"
#define CHIP_DEVICE_CONFIG_DEVICE_VENDOR_NAME "vendor test"
#define CHIP_DEVICE_CONFIG_DEVICE_PRODUCT_NAME "product test"
#define CONFIG_CHIP_DEVICE_PRODUCT_NAME "product name test"
#define CHIP_DEVICE_CONFIG_DEVICE_HARDWARE_VERSION_STRING "v1.0"
#define CHIP_DEVICE_CONFIG_DEFAULT_DEVICE_HARDWARE_VERSION_STRING "v2.0"

static const char *TAG = "app_main";
uint16_t light_endpoint_id = 0;

using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;
using namespace chip::app::Clusters;

constexpr auto k_timeout_seconds = 300;

#if CONFIG_ENABLE_ENCRYPTED_OTA
extern const char decryption_key_start[] asm("_binary_esp_image_encryption_key_pem_start");
extern const char decryption_key_end[] asm("_binary_esp_image_encryption_key_pem_end");

static const char *s_decryption_key = decryption_key_start;
static const uint16_t s_decryption_key_len = decryption_key_end - decryption_key_start;
#endif // CONFIG_ENABLE_ENCRYPTED_OTA

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
    case chip::DeviceLayer::DeviceEventType::kInterfaceIpAddressChanged:
        ESP_LOGI(TAG, "Interface IP Address changed");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningComplete:
        ESP_LOGI(TAG, "Commissioning complete");
        break;

    case chip::DeviceLayer::DeviceEventType::kFailSafeTimerExpired:
        ESP_LOGI(TAG, "Commissioning failed, fail safe timer expired");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStarted:
        ESP_LOGI(TAG, "Commissioning session started");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStopped:
        ESP_LOGI(TAG, "Commissioning session stopped");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowOpened:
        ESP_LOGI(TAG, "Commissioning window opened");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowClosed:
        ESP_LOGI(TAG, "Commissioning window closed");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricRemoved:
        ESP_LOGI(TAG, "Fabric removed successfully");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricWillBeRemoved:
        ESP_LOGI(TAG, "Fabric will be removed");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricUpdated:
        ESP_LOGI(TAG, "Fabric is updated");
        break;

    case chip::DeviceLayer::DeviceEventType::kFabricCommitted:
        ESP_LOGI(TAG, "Fabric is committed");
        break;
    default:
        break;
    }
}

static esp_err_t app_identification_cb(identification::callback_type_t type, uint16_t endpoint_id, uint8_t effect_id,
                                       uint8_t effect_variant, void *priv_data)
{
    ESP_LOGI(TAG, "Identification callback: type: %u, effect: %u, variant: %u", type, effect_id, effect_variant);
    return ESP_OK;
}

static esp_err_t app_attribute_update_cb(attribute::callback_type_t type, uint16_t endpoint_id, uint32_t cluster_id,
                                         uint32_t attribute_id, esp_matter_attr_val_t *val, void *priv_data)
{
    esp_err_t err = ESP_OK;

    if (type == PRE_UPDATE) {
        /* Driver update */
        app_driver_handle_t driver_handle = (app_driver_handle_t)priv_data;
        err = app_driver_attribute_update(driver_handle, endpoint_id, cluster_id, attribute_id, val);
    }

    return err;
}

static esp_err_t on_attribute_update(attribute::callback_type_t type, uint16_t endpoint_id, uint32_t cluster_id,
                                     uint32_t attribute_id, esp_matter_attr_val_t *val, void *priv_data) {
  return ESP_OK;
}
extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();
    // Setup Matter node
  node::config_t node_config;
  node_t *node = node::create(&node_config, on_attribute_update, app_identification_cb);

  // Set product name (stored in node 0)
  auto endpoint = esp_matter::endpoint::get(node, 0);
  auto cluster = esp_matter::cluster::get(endpoint, chip::app::Clusters::BasicInformation::Id);
  auto attribute = esp_matter::attribute::get(cluster, chip::app::Clusters::BasicInformation::Attributes::NodeLabel::Id);
  std::string data = CHIP_DEVICE_CONFIG_PRODUCT_NAME;
    char* nonConstData = new char[data.length() + 1];
    std::strcpy(nonConstData, data.c_str());
    auto val = esp_matter_char_str(nonConstData, data.length());
    delete[] nonConstData;

  esp_matter::attribute::set_val(attribute, &val);
    // Add enpoint to the node
  thermostat::config_t thermostat_config;
  thermostat_config.thermostat.local_temperature = 19;
  thermostat_config.thermostat.system_mode        = 0;    // OFF state
  endpoint_t *thermostat_endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);

  uint16_t thermostat_endpoint_id = endpoint::get_id(thermostat_endpoint);
  ESP_LOGI(TAG, "Thermostat endpoint_id = %"PRIu32, static_cast<uint32_t>(thermostat_endpoint_id));

  // Add additional feature
  cluster_t *thermostat_cluster = cluster::get(thermostat_endpoint, Thermostat::Id);
  cluster::thermostat::feature::heating::config_t heating_config;
  heating_config.abs_max_heat_setpoint_limit = 3000;
  heating_config.abs_min_heat_setpoint_limit = 1500;
  heating_config.max_heat_setpoint_limit = 2800;
  heating_config.min_heat_setpoint_limit = 2000;
  heating_config.occupied_heating_setpoint = 2300;
  heating_config.pi_heating_demand = 0;
  ESP_LOGE(TAG, "Thermostat heating cluster id = %"PRIu32, cluster::thermostat::feature::heating::get_id());
  cluster::thermostat::feature::heating::add(thermostat_cluster, &heating_config);

  // Add additional cooling cluster
//   cluster_t *thermostat_cluster = cluster::get(thermostat_endpoint, Thermostat::Id);
  esp_matter::cluster::thermostat::feature::cooling::config_t cooling_config;
  //cooling_config.config();MutableCharSpan
  cooling_config.abs_min_cool_setpoint_limit = 1600;
  cooling_config.abs_max_cool_setpoint_limit = 3200;
  cooling_config.pi_cooling_demand           = 50;  // 50%
  cooling_config.occupied_cooling_setpoint   = 2600;
  cooling_config.min_cool_setpoint_limit     = 1600;
  cooling_config.max_cool_setpoint_limit     = 3200;
  ESP_LOGE(TAG, "Thermostat cooling cluster id = %"PRIu32, esp_matter::cluster::thermostat::feature::cooling::get_id());
  esp_matter::cluster::thermostat::feature::cooling::add(thermostat_cluster, &cooling_config);

#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
    /* Set OpenThread platform config */
    esp_openthread_platform_config_t config = {
        .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(),
        .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),
    };
    set_openthread_platform_config(&config);
#endif

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

#if CONFIG_ENABLE_ENCRYPTED_OTA
    err = esp_matter_ota_requestor_encrypted_init(s_decryption_key, s_decryption_key_len);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialized the encrypted OTA, err: %d", err);
    }
#endif // CONFIG_ENABLE_ENCRYPTED_OTA

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#endif
}

```

This exposes 3 interfaces 
[image]

And in the home app i see this:

[image]

jadhavrohit924, 2023-08-24:
Hi @beckmx, device is properly working on my side. UI showed device not responding for few seconds and it disappeared and I could able to control thermostat using Apple home. Could you please try again and tell me your Apple home version?
Mine is 16.6

[image]

beckmx, 2023-08-24:
I use 16.2, could that be it?

jadhavrohit924, 2023-08-25:
Could be, please update and try again.

beckmx, 2023-09-02:
hello @jadhavrohit924 i updated the phone and yes, it started to show the controls correctly, the temperature shows as 0 degrees, what example do you think i could use to report the temperature and also, i wanted to ask you if there is an example to catch the requested temperature, from my code i am not sure how i can get the requested temp

thanx!

jadhavrohit924, 2023-09-04:
There is [`TemperatureSensor`](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/device_types/TemperatureSensor.adoc) device type that you can use to report the temperature. There is not any app for now to do so.

beckmx, 2023-09-06:
I added the temperature sensor but there is something wierd, it still shows 0.0 degrees, and only below the main title, appears in subtitle the temperature from the sensor

jadhavrohit924, 2023-09-06:
Could you please add the screen shots?

beckmx, 2023-09-06:
Here are three images of the different stages, when is off, when on and when you open the window detail [image]
[image]
[image]

jadhavrohit924, 2023-09-22:
@beckmx Sorry for the delayed response. You are right. We are observing same behaviour. We will check with the Apple and update.

jadhavrohit924, 2024-02-21:
@beckmx I have tried this with apple, since the temperature in matter have resolution of 0.01°C, if the measured temperature value is 20 or 30 i.e 0.2°C  or 0.3°C, respectively. It is relatively very less. If you set the measured temperature value to lets say 200, then apply home will show it as 2°C. Please try it out and close the issue if it works.
[image]

jadhavrohit924, 2024-08-09:
@beckmx Please close the issue if it is resolved.

dhrishi, 2024-10-15:
Closing this now

closed 2024-10-15 by dhrishi