espressif/esp-matter #588: Matter Controller example: Can not write commands

nguyenhuy5997, 2023-08-24:
esp-mater branch: main
esp-idf branch: v5.0.1
module: esp32-s3
Flash and run controller examples, i can not write commands over serial
here is the log

I (1552) NimBLE: GAP procedure initiated: stop advertising.

I (1552) chip[DL]: CHIPoBLE advertising stopped
I (1562) chip[DL]: HandlePlatformSpecificBLEEvent 32779
E (1562) app_main: Matter start failed: 259
> I (1882) chip[SPT]:  ----- Commissioner using vendorId 0xFFF1
I (1882) chip[SVR]: Retrieving value from controller server storage.
I (1892) chip[SVR]: Retrieved value from server storage.
I (1892) chip[SVR]: Retrieving value from controller server storage.
I (1902) chip[SVR]: Retrieved value from server storage.
I (1912) chip[SPT]:  ----- UDC listening on port 5560
I (1972) chip[SVR]: Retrieving value from controller server storage.
I (1972) chip[SVR]: Retrieved value from server storage.
I (1972) chip[SVR]: Retrieving value from controller server storage.
I (1992) chip[SVR]: Retrieved value from server storage.
I (1992) chip[CTL]: Generating NOC
I (2062) chip[SVR]: Retrieving value from controller server storage.
I (2062) chip[SVR]: Retrieved value from server storage.
I (2062) chip[SVR]: Retrieving value from controller server storage.
I (2082) chip[SVR]: Retrieved value from server storage.
I (2082) chip[SVR]: Stored value in server storage
I (2102) chip[SVR]: Stored value in server storage
I (2102) chip[ZCL]: Using ZAP configuration...
I (2102) esp_matter_cluster: Cluster plugin init common callback
I (2122) chip[FP]: Validating NOC chain
I (2582) chip[FP]: NOC chain validation successful
I (2582) chip[FP]: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
I (2582) chip[TS]: Last Known Good Time: 2023-08-25T00:29:49
I (2602) chip[TS]: New proposed Last Known Good Time: 2021-01-01T00:00:00
I (2602) chip[TS]: Retaining current Last Known Good Time
I (2612) chip[SVR]: Retrieving value from controller server storage.
I (2622) chip[FP]: Metadata for Fabric 0x1 persisted to storage.
I (2632) chip[TS]: Committing Last Known Good Time to storage: 2023-08-25T00:29:49
I (2652) chip[CTL]: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 55427473FA54B5E5)
I (2652) chip[CTL]: *** Missing DeviceAttestationVerifier configuration at DeviceCommissioner init: using global default, consider passing one in CommissionerInitParams.
I (2672) chip[SPT]: Setting up group data for Fabric Index 1 with Compressed Fabric ID:
I (2672) chip[SVR]: Retrieving value from controller server storage.
I (2692) chip[SVR]: Retrieved value from server storage.
I (2692) chip[SVR]: Retrieving value from controller server storage.
I (2702) chip[SVR]: Retrieved value from server storage.
I (2702) chip[SVR]: Stored value in server storage
I (2712) chip[SPT]: InitCommissioner nodeId=0x000000000001B669 fabricIndex=0x1
I (2722) chip[DL]: HandlePlatformSpecificBLEEvent 53255
I (2722) chip[DL]: HandlePlatformSpecificBLEEvent 53254

wqx6, 2023-08-28:
Are you using the UART0 as the console channel? or USB/JTAG? If you are using `USB Serial/JTAG Controller`, you should select `ESP_CONSOLE_USB_SERIAL_JTAG` in menuconfig `→ Component config → ESP System Settings → Channel for console output`.

closed 2023-08-28 by nguyenhuy5997

DanTup, 2023-12-07:
Aah, that makes sense - thank you! :-)