espressif/esp-matter #589: Heap/Ram memory requirement for ESP-Matter SDK in ESP32C3

aggaddam, 2023-08-25:
My product is using esp-matter sdk and also runs the HTTPS/MQTTS task on top of esp -matter sdk once device is up on the wifi network.  From my initial testing, i observed that running chip sdk and HTTPS/MQTTS tasks causes shrinking of free available heap memory to lowest levels i.e ~15K - 20K free memory. 

I just want to understand the memory requirement of RAM/Heap memory for CHIP SDK.

shubhamdp, 2023-08-30:
@aggaddam Our examples has [device performance](https://github.com/espressif/esp-matter/tree/main/examples/light#3-device-performance) section which mentions the heap numbers. I know they are pretty old now, will update with the newer ones, but I do not anticipate a lot of increase in RAM consumption.

jonsmirl, 2023-09-18:
The C3 has enough memory to run Matter, but it does not have enough memory to run much of anything else in addition to Matter.  I got tired of fighting out-of-memory on C3 and switched over to the S3-N4R2-MINI. It is only a little more expensive and now I have plenty of memory due to the 2MB of PSRAM. It is trivial to switch, just recompile and adjust the GPIO numbers.

dhrishi, 2023-12-04:
Closing as the question is answered