espressif/esp-matter #590: Question - Can i build multiple output?

mrezzaaa, 2023-08-25:
I have question about esp32 matter device to work with with multiple GPIO output. I had trying to make multiple endpoint with each endpoint is responding to each GPIOs. Testing with CHIP-Tools in ios, it have multiple selection with light endpoint 1-10 and run like expected (controlling relay channel from channel 1 to 8). 
But when using Tuya only detected single endpoint such as relay on channel 1 only.
Also trying with ESP Matter app in iOS only endpoint 1 detected.

I think matter can support single devices with multiple output, what im gonna to do? I don't know how to figure it out. And now im stucked :(.

jonsmirl, 2023-08-27:
This should work as expected with each GPIO corresponding to an OnOff endpoint.  I know it will work in the Apple and Google implementations. I don't work with Tuya.  Did you set the device type for OnOff on each of the endpoints?

mrezzaaa, 2023-08-27:
```
for(int i=0;i < relays.size();i++){
        ESP_LOGW(TAG,"Preparing relay channel  %i",relays.at(i));
        app_driver_handle_t light_handle = app_driver_light_init(relays.at(i));
        on_off_light::config_t light_config;
        light_config.on_off.on_off = DEFAULT_POWER;
        endpoint = on_off_light::create(node,&light_config,ENDPOINT_FLAG_NONE,light_handle);
        light_endpoint_id = endpoint::get_id(endpoint);
        endpoints.push_back(light_endpoint_id);
        ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);
        if (!node || !endpoint) {
            ESP_LOGE(TAG, "Matter node creation failed");
        }
    }
```
i guest i did, right ?

shripad621git, 2023-08-30:
@mrezzaaa  you can have a look at [genric_switch_example](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp) where we have made a provision to map gpio to endpoint for reference.

mrezzaaa, 2023-08-30:
@shripad621git I've done tried as your suggestion. But it only scanned only 1 endpoint. Actual i have 5 light endpoint.

Check my log:
```
I (31) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (31) boot: compile time Aug 30 2023 21:06:46
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (77) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (84) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
I (92) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (99) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (106) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (114) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (122) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (129) boot: End of partition table
I (133) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=392a8h (234152) map
I (190) esp_image: segment 1: paddr=000592d0 vaddr=3ff80000 size=00004h (     4) load
I (190) esp_image: segment 2: paddr=000592dc vaddr=3ffbdb60 size=056f4h ( 22260) load
I (201) esp_image: segment 3: paddr=0005e9d8 vaddr=40080000 size=01640h (  5696) load
I (206) esp_image: segment 4: paddr=00060020 vaddr=400d0020 size=fcdfch (1035772) map
I (426) esp_image: segment 5: paddr=0015ce24 vaddr=40081640 size=1d858h (120920) load
I (474) boot: Loaded app from partition at offset 0x20000
I (474) boot: Disabling RNG early entropy source...
I (485) cpu_start: Multicore app
I (486) cpu_start: Pro cpu up.
I (486) cpu_start: Starting app cpu, entry point is 0x400814a4
0x400814a4: call_start_cpu1 at /Users/icewalker/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (506) cpu_start: Pro cpu start user code
I (506) cpu_start: cpu freq: 160000000 Hz
I (506) cpu_start: Application information:
I (510) cpu_start: Project name:     light_relay
I (516) cpu_start: App version:      1.0
I (520) cpu_start: Compile time:     Aug 30 2023 21:06:40
I (526) cpu_start: ELF file SHA256:  33bc50b1a4972beb...
I (532) cpu_start: ESP-IDF:          v5.1.1
I (537) cpu_start: Min chip rev:     v0.0
I (542) cpu_start: Max chip rev:     v3.99 
I (547) cpu_start: Chip rev:         v3.0
I (552) heap_init: Initializing. RAM available for dynamic allocation:
I (559) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (565) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (571) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (577) heap_init: At 3FFD4BD8 len 0000B428 (45 KiB): DRAM
I (583) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (590) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (596) heap_init: At 4009EE98 len 00001168 (4 KiB): IRAM
I (604) spi_flash: detected chip: generic
I (607) spi_flash: flash io: dio
I (614) coexist: coex firmware version: 80b0d89
I (617) app_start: Starting scheduler on CPU0
I (621) app_start: Starting scheduler on CPU1
mI (621) main_task: Started on CPU0
I (631) main_task: Calling app_main()
I (771) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (771) esp_matter_core: Attribute 0x0000FFFD on cluster 0x0000001D already exists. Not creating again.
W (781) esp_matter_core: Attribute 0x0000FFFD on cluster 0x0000001F already exists. Not creating again.
E (791) esp_matter_attribute: Could not create attribute, string length out of bound
E (801) esp_matter_attribute: Could not create attribute, string length out of bound
W (811) esp_matter_core: Attribute 0x00000000 on cluster 0x00000030 already exists. Not creating again.
W (821) app_main: Creating button 0 with GPIO 13
[... 209 lines cut ...]
I (4391) esp_netif_handlers: sta ip: 192.168.100.14, mask: 255.255.255.0, gw: 192.168.100.1
I (4391) chip[DL]: IP_EVENT_STA_GOT_IP
I (4391) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.100.14/255.255.255.0 gateway 192.168.100.1
I (4401) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (4421) app_main: Interface IP Address changed
I (4421) chip[DIS]: Updating services using commissioning mode 0
I (4431) chip[DIS]: CHIP minimal mDNS started advertising.
I (4451) chip[DIS]: Advertise operational node 5070A2F460AE7060-0000000000000075
I (4451) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5070A2F460AE7060-0000000000000075.
I (4471) chip[DIS]: mDNS service published: _matter._tcp
I (4471) chip[DIS]: Advertise operational node 906D89248D0488D1-0000000100000014
I (4481) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 906D89248D0488D1-0000000100000014.
I (4501) chip[DIS]: mDNS service published: _matter._tcp
I (4501) chip[DIS]: Advertise operational node 2A893D1B8C8F99AB-0000000000000047
I (4511) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 2A893D1B8C8F99AB-0000000000000047.
I (4521) chip[DIS]: mDNS service published: _matter._tcp
I (4531) chip[DIS]: Checking node lookup status after 338 ms
```
[image]
[image]
[image]
[image]

On Chip-Tool always work but must be choosing light endpoint manually. 
I'm testing using 3 apps ( work without gateway : Tuya Smart, Espressif Matter and CHIP-Tool ).

shripad621git, 2023-08-31:
@mrezzaaa, have you done some changes to the app_driver.cpp code wherein you address the light_endpoint_id variable which is extern.
You can have a closer look at the generic_switch [app_driver.cpp](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_driver.cpp) wherein the same issue was addressed to support multiple endpoints.
You will need some provision to identify the endpoint for which the onoff or toggle command is triggered in the toggle callback in app_driver.cpp of the light example.

mrezzaaa, 2023-08-31:
@shripad621git like function ``app_driver_button_initial_pressed`` you mean ?

shripad621git, 2023-09-04:
Yes, you can give it a try. You can refer [these](https://github.com/espressif/esp-matter/pull/330/commits/7f97c973117550bbcd8f590afdcd3bf3780551fb) changes to do so.

shripad621git, 2023-09-06:
@mrezzaaa  any update on the issue?Please close the issue if it is solved.

mrezzaaa, 2023-09-06:
Not yet, still working on that. I'll close this issue if it works.

dhrishi, 2023-09-21:
@mrezzaaa Please close the issue if this works.

closed 2023-09-22 by mrezzaaa