espressif/esp-matter #591: Failed to get matter code

tigertclhu, 2023-08-26:
fatal: remote error: upload-pack: not our ref ef05eb60565ce3be035b79b8fac1f9d12c7e706b
fatal: Fetched in submodule path 'connectedhomeip/connectedhomeip/third_party/silabs/gecko_sdk', but it did not contain ef05eb60565ce3be035b79b8fac1f9d12c7e706b. Direct fetching of that commit failed.

tigertclhu, 2023-08-26:
this commit id "ef05eb60565ce3be035b79b8fac1f9d12c7e706b" cannot be found in this link "https://jihulab.com/esp-mirror/SiliconLabs/gecko_sdk.git"

dhrishi, 2023-08-26:
@tigertclhu The submodule on the latest main branch is https://github.com/espressif/connectedhomeip/tree/5b4f8004662d00bdb111367fec7d3ea978c23372 and the submodule within it is also there: https://github.com/SiliconLabs/gecko_sdk/tree/ef05eb60565ce3be035b79b8fac1f9d12c7e706b 

It should work fine. Please check

dhrishi, 2023-09-14:
Closing this now

closed 2023-09-14 by dhrishi