espressif/esp-matter #592: "error: command 'x86_64-linux-gnu-gcc' failed: No such file or directory"

vishesh-varma, 2023-08-27:
I was following the "Developing with the SDK guide" on my Ubuntu WSL. I had a lot of errors come while I was trying to build it, but thankfully found a solution to all of them up until this point. Here's my output after running ./install.sh
Since the output is too long for an issue: here's the pastebin: https://pastebin.com/htguL32n

The message below is just the last snippet.
```
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_pss_2048_sha256_mgf1_32_test.json -> build/lib.linux-x86_64-cpython-310/Crypto/SelfTest/Signature/test_vectors/wycheproof
      running build_ext
      building 'Crypto.Hash._MD2' extension
      creating build/temp.linux-x86_64-cpython-310
      creating build/temp.linux-x86_64-cpython-310/src
      x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=64 -DLTC_NO_ASM -Isrc/ -I/home/vish/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/include -I/usr/include/python3.10 -c src/MD2.c -o build/temp.linux-x86_64-cpython-310/src/MD2.o
      error: command 'x86_64-linux-gnu-gcc' failed: No such file or directory
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycryptodome
ERROR: Could not build wheels for pycryptodome, which is required to install pyproject.toml-based projects
```

sayondeep, 2023-08-27:
@vishesh-varma Did you follow the pre-requisites for matter as mentioned [here](https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites)?

vishesh-varma, 2023-08-27:
Totally skipped the matter prerequisites part. I got it working eventually, installing one library at a time as things broke, but that I am going through the commands linked, that pretty much handle all of my dependencies. 
Thanks though!

closed 2023-08-27 by vishesh-varma